mc0110/inetbox2mqtt #6: WLAN/MQTT instabil, Topic display_status

Harry-58, 2022-11-07:
Hallo,
ich hab jetzt die inetbox seit 2 Tagen am laufen. 
Hat auch sofort funktioniert.  Tolle Arbeit!!!

Mir sind aber 2 Dinge aufgefallen
1. Im Originalprogramm von Daniel Fett gibt es auch das Topic " service.truma.display_status", dass ich hier vermisse.
2. Bei mir ist die Wlan-Verbindung recht instabil wegen großem Abstand zwischen WoMo und MQTT-Server.
    Dadurch wird die Verbindung häufig unterbrochen.
    Leider sieht man das am ESP nicht, da die LED (MQTT) immer an bleibt, wenn einmal eine Verbindung bestand.
    
    Deshalb habe ich truma_serv.py folgendermaßen erweitert:

Neue Funktion, damit die LED bei Unterbrechung der Verbindung ausgeschaltet wird.
```
async def wifi_callback(status):
    print(f"Wifi-Status: {status}")
    if not status: set_led("MQTT", False)
```

Funktion erweitert, damit die LED bei Wiederherstellung der Verbindung eingeschaltet wird.
```
async def conn_callback(client):
    set_led("MQTT", True)
    await client.subscribe(S_TOPIC_1+"#", 1)
    await client.subscribe(S_TOPIC_2, 1)
```

config für mqtt erweitern, damit die neue Funktion bei Änderung des Verbindungsstatus aufgerufen wird.
```
config.subs_cb  = callback
config.connect_coro = conn_callback  #notification when a MQTT connection starts
config.wifi_coro = wifi_callback     #notification when wifi connects/disconnects
```

Wäre schön, wenn das in die nächste Version übernommen werden könnte.

mc0110, 2022-11-07:
Hello,
I'm very pleased that the installation worked for you. I assume this is a test scenario where your wifi signal is weak, right?
- status messages: This is implemented in the code, but my CPplus doesn't send 0x20-0x22 messages, so I don't have any way of testing this and have to rely on help here.
- wifi status: Your suggestion is a good one, I'll take a closer look at it.

P.S. I hope you are comfortable with us keeping it in English here on Github.

Harry-58, 2022-11-07:
Hello,
thanks for the quick reply.
The camper is in front of my apartment and connects to my Smarthome (IoBroker)
Therefore, I have only very weak WiFi.
I don't have a WiFi access point inside the camper.
Unfortunately, I only have basic knowledge of Python, so I can only provide limited support with regard to 0x20-0x22 messages.
My world is rather C++ and C#.
Nevertheless, I am trying to extend the inetbox with a SMS interface (SIM800L).
If it works, I can make it available.

mc0110, 2022-11-09:
The functionality is built into the next release - I would therefore close the issue.

closed 2022-11-09 by mc0110