mc0110/inetbox2mqtt #13: Connection Schema for Duo Control

wolfganghuse, 2022-11-20:
Hello, will you also provide a connection schema for Duo Control ? 

Thanks for this nice implementation on ESP32.

P.S.: Would be nice to keep this clean from other not inetbox-related features.

mc0110, 2022-11-20:
Hi Wolfgang,

I am thinking about how we can better separate the different features. However, it is manageable at the moment. In the code, the segments in truma_serv.py can be

   #Config Features
    activate_duoControl = True
    activate_spiritlevel = True

can be activated / deactivated respectively. If an element is set to False, no code is executed for it.

I'll try to concretise the details of the connection schema even further.

wolfganghuse, 2022-11-20:
Yes, understood... i am not using HomeAssistent but VenusOS and therefore i need another kind of mqtt-integration... i will use the the same approach `activate_dbus_mqtt_devices = True` in my fork.
Maybe you can also put the HomeAssistent Config into segments ?

mc0110, 2022-11-20:
good point!
It would be great if you could show examples of how you did the implementation. I would like to expand the point and include examples for different systems.

wolfganghuse, 2022-11-20:
I am using this project on my Venus to easily add new services through mqtt:
https://github.com/freakent/dbus-mqtt-devices
There is Register-Process which does all the internal logic

A simple implementation can be found in my Repo where i collect temperature via BLE and pass it via mqtt:
https://github.com/wolfganghuse/ble-mqtt

wolfganghuse, 2022-11-30:
I guess LevelControl is not using BLE GATT Services which are quite easy to handle... if there is some proprietary protocol or even encrypted traffic it fast gets ugly.
Same with Mopeka Gas Level Sensors, the cheap simple ones have a quite complicated protocol but their Pro-Version is using easy BLE Advertising.
Do you have access to a LevelControl to collect some BLE Traffic ? (You can also reach out to me via Telegram-Group)

nightcat91, 2022-12-03:
Hi,

@7wells and @mc0110, sorry for my long time away and no communication... first I was sick and then had a lot of work to catch up with, so no time for tinkering and the camper!

But I was able to get the software running on my ESP32 today using the flash via ESPtools and then the setup via Putty. 
So far it seems to work fine, had to flash a second time cause I enable the spiritlevel feature and did not have a MPU6050 connected...
I will try and connect everything with the Truma in the next days!

On the topic of DuoControl:
I have a Duo Control fitted in my camper, but I dont really understand the connection nor the feature of the in this software here. What is the purpose of the 2 inputs and 2 outputs?
How would I go about connecting them (I never pulled the DuoControl out of the wall yet)?
And what can I do with it then?

@mc0110 Thanks again for your great work man! The new Readme was very easy to understand and the install was rather simple using the prebuild file and ESPTools.
Maybe a bit of explanation about the DuoControl "addon" in the readme would make it a bit more clear.

mc0110, 2023-01-03:
Hi,

I have added details about the DuoControl to the README. I hope it is OK for you. Unfortunately I didn't get more information about the system and I don't have one myself. I don't know if anyone uses it. But you can also control completely different things with my wiring suggestion.

closed 2023-01-03 by mc0110