mc0110/inetbox2mqtt #14: Function of dtoggle_led in tools.py ???

Harry-58, 2022-11-24:
What should the function "dtoggle_led" in tools.py do?
Two times NOT results in YES, which does not change the state of the LED!

mc0110, 2022-11-24:
This is exactly, what we want. So the state won't changed but the led flickers. Pls see the [README](https://github.com/mc0110/inetbox2mqtt#esp32-gpios-and-leds---debugging-will-be-easier) for details.

closed 2022-11-24 by mc0110

Harry-58, 2022-11-24:
I think that is wrong.
The two lines are without delay directly behind each other.
The time interval is only nano/microseconds.
There is absolutely nothing to see on the LED!
I have tested this in a loop.

mc0110, 2022-11-24:
Sorry, Harry - the code is running and I see the led flickering. I agree, that I wasn't sure, that it will be function without delay, but it do!

Harry-58, 2022-11-24:
I think the flickering you see comes from the toggle_led in _send_answer.

mc0110, 2022-11-24:
If you think so - without INIT on CPplus the port doesn't send one byte, but the led is flickering. I don't understand, what the problem is because it is only a debugging feature.