mc0110/inetbox2mqtt #16: MPU6050

maseb24, 2022-11-30:
Hallo Magnus,

ich habe deinen code mit einem ESP32 schon am laufen. Alles ok soweit.

Nun habe ich versucht deine Wasserwaage zu installieren. Bisher habe ich immer dein .bin file mit esptool auf den ESP32 kopiert.
Nur hat das aktuelle .bin file keine unterstützung für den MPU6050. Dann habe ich die fehlenden .py files mit Thonny auf den ESP32 kopiert.  Nach einem reboot sehe ich in Thonny beim starten des ESP32 nur noch unleserliche Zeichen. 

Kannst du mir einen Tipp geben was ich falsch mache.

Gruß Rudi

mc0110, 2022-11-30:
Hello Rudi,
I hope you understand that we want to keep this issue discussion in English so that everyone can participate. But you are right, I have not included the MPU6050 integration in the .bin file. I will adjust that in the short term. Please excuse that!
I have not yet been able to observe the behaviour with Thonny. In this case I recommend reinstalling a Python image (in Thonny in the "RUN" menu) and then reinstalling the sources from the clone of the repo.
I hope that helps!

maseb24, 2022-11-30:
Hello Magnus,

Thanks for the great project.
I copied the micropython "esp32-20220618-v1.19.1" to my ESP32, then all the other .py files.
Unfortunately the result stayed the same. Garbled strings after startup.

Thanks for the help Rudi

mc0110, 2022-11-30:
Hey Rudi,
I'm gonna need a little more to go on. What exactly is not working?

maseb24, 2022-11-30:
Am Wed, 30 Nov 2022 12:16:05 -0800
schrieb Magnus ***@***.***>:

> Hey Rudi,
> I'm gonna need a little more to go on. What exactly is not working?
> 

Hallo Magnus,

ich nehme an in der email ist es nicht nötig den Text vorher durch den
Translator zu schicken.

Ich habe eben dein aktuelles .bin file auf meinen ESP32 geflasht.
Vorher habe ich diesen mit  blank4mb.bin gelöscht.

Das Ergebnis ist leider wie vorher. Die inetbox bricht mit einerm
Fehler ab. In Thonny, in der shell erscheint folgendes direkt nach dem
starten.

"
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12344
ho 0 tail 12 room 4
load:0x40080400,len:4124
entry 0x40080680
Check for credentials.dat
credentials.dat file exists -> pass the ask for credentials
Found ESP32 Board, using UART2 on GPIO 16 and
1�����������������������������������������������������������������������������������������������������������������"

Es kommt keine Verbindung mit mqtt zustande.

Mir ist noch aufgefallen das mit der Version 0.8.4 von truma_serv.py,
die inetbox ohne Fehler startet

LG Rudi

mc0110, 2022-11-30:
Hi Rudi,
the good news are - I can reproduce your error - see the same picture.
The error happens, if you activate the spiritlevel-part. So the easiest workaround is to deactivate this part of the software.
I set there a new version of the image (with deactivated spiritlevel) and hope, skrebber can help - unfortantely this part of the software isn't my own. This was the bad news.

mc0110, 2022-12-01:
Hi @maseb24

the error could be clarified. Unfortunately, the I2C pins for the SpiritLevel support were chosen in an unsuitable way and therefore no monitor was possible after the initialisation of the I2C bus. Please use the new version that has just been posted. New in this version is that the desired addons are queried in the start menu. Sadly it is unavoidable to connect the MPU6050 to the changed pins. Sorry again for this inconvenience.

maseb24, 2022-12-01:
Hello Magnus,

I just flashed your current version on my ESP32. I only chose the spirit level as a selection.
After rebooting the ESP32, it aborts with the following error message.
I checked the wiring several times SDA to GPIO26 and SCL to 25.

"Found ESP32 Board, using UART2 for LIN on GPIO 16(rx), 17(tx)
Activate spirit_level set to true, using I2C- on GPIO 25(scl), 26(sda)
Traceback (most recent call last):
  File "main.py", line 6, in <module>
  File "truma_serv.py", line 74, in <module>
  File "spiritlevel.py", line 39, in __init__
  File "imu.py", line 95, in __init__
MPUException: No MPU's detected"

Greetings Rudi

skrebber, 2022-12-04:
At the moment I have fixed the MPU board with a Tesa PowerStrip behind the Truma CPplus. In my Hymer there is enough room for the stepdown converter, the uC, Lin transceiver board and the MPU. 

Noise should not be a problem thanks to the Kalman filter. And in my installation it is definitely not.

closed 2023-01-03 by mc0110