mc0110/inetbox2mqtt #18: ESP32 does not connect to WIFI after settings confidential

snipah, 2022-11-30:
I programmed the ESP32 using all suggested methods and it does not connect to WIFI after setting and saving/ encrypting the WIFI/MQTT-Data.

Check for credentials.dat
Fill in your credentials:
SSID: poessl.net

Wifi-password: PassworT@ZUhause

MQTT-Server - IP or hostname: 192.168.1.21

Username: snipah

User-Password: anho1572

Your inputs are:

SSID       : poessl.net
Wifi-PW    : PassworT@ZUhause
MQTT-Server: 192.168.1.21
Username   : snipah
User-PW    : anho1572

ok for you (yes/no): yes
Write credentials encrypted to disk
Found ESP32 Board, using UART2 on GPIO 16 and 1�����������������������������������������������������������������������������������������������������������������

mc0110, 2022-11-30:
Sorry, I think you fetched a buggy version. I try to set a new version - at the moment the spiritlevel-part isn't running. I apologize for the situation.

mc0110, 2022-12-01:
Hi,
the error could be clarified. Unfortunately, the I2C pins for the SpiritLevel support were chosen in an unsuitable way and therefore no monitor was possible after the initialisation of the I2C bus. Please use the new version that has just been posted. New in this version is that the desired addons are queried in the start menu. Sadly it is unavoidable to connect the MPU6050 to the changed pins. Sorry again for this inconvenience.

snipah, 2022-12-01:
Thanks a lot for the quick fix! Tested and confirmed to be working!

mc0110, 2022-12-01:
Really good news - thanks for your feedback!

snipah, 2022-12-02:
New version boots up nicely.

closed 2022-12-02 by snipah