mc0110/inetbox2mqtt #20: Connection not working!

snipah, 2022-12-01:
Check for credentials.dat
credentials.dat file exists -> pass the ask for credentials
Found ESP32 Board, using UART2 for LIN on GPIO 16(rx), 17(tx)
main-loop is running
INFO:mqtt_async:connecting wifi
lin-loop is running
INFO:mqtt_async:Connecting to ('192.168.1.21', 1883) id=b'0cb815cd8ff4' clean=1
INFO:mqtt_async:Connecting to ('192.168.1.21', 1883) id=b'0cb815cd8ff4' clean=0
Wifi connected
MQTT connected
set ha_autoconfig
publish energy_mix:gas
publish el_power_level:0
publish target_temp_water:0
publish target_temp_room:0
publish alive:OFF
publish heating_mode:off
publish operating_status:Off
publish current_temp_room:0
publish current_temp_water:0
publish alive:OFF
publish alive:OFF
publish alive:OFF

What could I check? I switched TX/RX from the ESP and also LIN/Ground. Does not seem to connect. Also did several resets on the CP+. No clue! Installed the latest unstable MicroPython and set up everything again... no dice!

What could I check next....?

snipah, 2022-12-01:
[image]
[image]
[image]
[image]

mc0110, 2022-12-01:
The Python-Version is okay, with a older version it throws a error. The MQTT-Connection is fine. 
Tx (17) has to be connected to RX(Lin-Interface). I think, this is wrong. You should use the LED-Debugger. In my car with my first tries (flying connections) I killed one LIN-UART and also one RPI pico (is running, but the tx-GPIO is quiet). After using the LED you see, if the Interface is sending anything or if it is dead.

snipah, 2022-12-01:
Switched RX and TX.  No dice! What do you mean with grouding?

snipah, 2022-12-01:
> Tx (17) has to be connected to RX(Lin-Interface). I think, this is wrong. 

The readme says differently..

snipah, 2022-12-02:
Ordered a replacement LIN-UART and will send the old one back, as I assume it was not working correctly out of the box. Also ordered an MPU6050. Lets see how this can nicely be integrated into Home Assistant. BTW: Thanks for your efforts ;)

snipah, 2022-12-02:
> I also ordered an MPU6050. If you're interested, there's a separate "issue" (#16) about MPU6050-related topics. 😉

...then lets work on something nice for home assistant. BTW: Ich bin "snipah" im Pösslforum.

mc0110, 2022-12-02:
Hey snipah,
I am sorry that you were not successful.  But you are not the first to have problems with the LIN-TTL module.
I'll think about more diagnostics in the future.

mc0110, 2022-12-02:
> Lets see how this can nicely be integrated into Home Assistant.

There is an example for a HA-integration from Sönke (skebber) in our docs

mc0110, 2022-12-05:
Hey snipah, I would close this issue, if it is okay for you. Hopefully you see a connection with the new device - your comments will be welcome

closed 2022-12-05 by mc0110

mc0110, 2022-12-06:
So you are right, on your pictures is clear to see the missing grounding - sorry that I didn't give you this hint. I'll try to pick this up in the docs

mc0110, 2023-02-15:
[image]

Here is another very clear connection diagram from a successful installation

tojo99, 2023-12-18:
Hi, I want ask you about project inetbox. Is that right connection? 
[image]
Between gnd and lin is still 1,5V (Uart Transceiver)? Thank you.

mc0110, 2023-12-18:
of course