mc0110/inetbox2mqtt #24: No CPplus INIT, LEDs Switched and Password with special Character

nightcat91, 2023-01-02:
Hi Magnus,

I finally came around to finish my whole PCB that is going into the camper and is also home to your amazing Port for controlling the Truma.
Today I test fitted everything and ran into the issue that I cant get a connection with Truma System.

Also while working the problem and using the LEDs to find the issue I ran into two other small bugs:
1.) The Debug LEDs seem to be switched around. The LED on GPIO14 was showing the MQTT Connection Status for me... 
2.) The software doesnt like it when the MQTT Password contains an exclamation mark  (!). I had that in my MQTT Password and the software kept coming back with the WifiManager and not going into the main program. After looking at the MQTT Password that is written out I noticed that it got a bit scrambled. After changing my MQTT Password that issue was gone.
I did not check any other special characters or if the same problems exists with the wifi password aswell, but those might be worth checking.

But back to my main issued of not getting any connection with the Truma System/CP Plus.
I am using and ESP32 and your newest .bin File V14.
I double checked all my connections and they seems OK, also when switching RX and TX between the LIN Bus Module and the ESP I get even less activity, so thats all fine. Also I have a common Ground for all devices!

I get a good startup of the program, connection to Wifi and MQTT is OK and the LED shows that, but never any connection to the Truma, but the second LED (on GPIO12 for me) is flickering every couple of seconds as described by you.
Also a Scan initiated from the CP Plus Panel doesnt do anything but show more activity on the Bus according to the LED.

This is the output log from the serial connection showing the startup. After that it just keeps on repeating the last line ("publish alive:OFF")

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4540
ho 0 tail 12 room 4
load:0x40078000,len:12344
ho 0 tail 12 room 4
load:0x40080400,len:4124
entry 0x40080680
Detected micropython 1.19.1 10502 on port: esp32
Connecting with credentials to network...
...STA connection connected successful
{'cred_fn': True, 'sta_hostname': 'ESP32TrumaControl', 'port': 'esp32', 'run_mode': 1, 'sta_reconnects': -1, 'sta_gateway': '192.168.111.1', 'sta_mask': '255.255.255.0', 'python': 'micropython 1.19.1 10502', 'sta_ip': '192.168.111.156', 'sta_ssid': 'wifionwheels', 'sta_mac': b"|\x9e\xbdI'\xf4", 'sta_state': 'on', 'sta_channel': 1, 'cred_bak': True, 'sta_txpower': 20.0, 'ap_state': 'off', 'sta_dns': '192.168.111.3'}
Normal mode activated - for chance to OS-mode type in terminal:
>>>import os
>>>os.remove('run_mode.dat'
{'SSID': 'wifionwheels', 'MQTT': '192.168.111.5', 'HOSTNAME': 'ESP32TrumaControl', 'ASL': '0', 'WIFIPW': '**************', 'ADC': '0', 'UN': '**********', 'UPW': '**********'}
Found ESP32 Board, using UART2 for LIN on GPIO 16(rx), 17(tx)
main-loop is running
INFO:mqtt_async:Connecting to ('192.168.111.5', 1883) id=b'7c9ebd4927f4' clean=1
INFO:mqtt_async:Connecting to ('192.168.111.5', 1883) id=b'7c9ebd4927f4' clean=0
Wifi connected
MQTT connected
set ha_autoconfig
lin-loop is running
publish energy_mix:gas
publish el_power_level:0
publish target_temp_water:0
publish target_temp_room:0
publish alive:OFF
publish heating_mode:off
publish operating_status:Off
publish current_temp_room:0
publish current_temp_water:0
publish alive:OFF
publish alive:OFF
publish alive:OFF
```

My green LED (GPIO14 for me) shows the MQTT Connection Status pretty quick but as described above the CPplus INIT never works.

Here you can find a small video showing the state of the LEDs: https://youtu.be/bdG-SAsgRDE
Also I think I have a newer version of CP Plus Panel, so attached are a few pictures of the panel and the Truma heater.

If you need any more info please let me know. We could also have a quick call (in German) to maybe discuss the problem directly.

Thanks for any help!

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mc0110, 2023-01-02:
Hi,

thxs for the detailed report.
 
To the small problems first. 
- I have extended the MQTT status, because when testing the web frontend it helps to have information about the status of the WiFi. I haven't included that in the readme yet though. Sorry for that.
- The problem with the special characters is new for me. I'll have a look at that.

Now to your major problem. The flickering of the LED looks OK, it's the same with me. I think the fault is not on the receiving side. You mqtt-connection is also okay. So it can only be on the transmitting side. Are you sure that the transmit channel is OK.

The CPplus panel really looks different from mine and also different from Daniel's picture. Is it a new heater? Allow me to ask, even though it may seem superfluous. You don't have an inetbox connected? The inetbox2mqtt does not work together with the Inetbox!

For further debugging:

- You can set debug_lin = True in truma_serv. Then you will get a much fuller activity report.
- You should check whether the ESP is really transmitting. I myself had the problem that one of my ports had a dead transmitting GPIO and spent two days looking for it. Please also consider the LIN-UART, there is obviously a lot of rejects - that would fit your error pattern.
- Which version does your Truma and your CPplus have?
- There is a Telegram group that I can recommend to you. There is a lot of expertise there.

Let's hold off complimenting the software until it's up and running for you. I am keeping my fingers crossed

mc0110, 2023-01-02:
Hi nightcat91,

I've had a look around on the Internet. You don't use CPplus! The hack will not work for you. The **inet x panel** includes the functions of inetbox and cpplus.

Sorry for the bad news

mc0110, 2023-01-02:
[image]

I asked on Telegram - sorry for the german answers

nightcat91, 2023-01-02:
Hi mc0110,

thanks for your quick reply and your research. I didnt look at the different CP Plus Panels that detailed....
I do not have a inetbox attached and I dont really wanne use one, cause in my opinion using SMS is very outdated when we can have a full HomeAssistant in the camper!!

I will have to do some more digging about the differences from the CP Plus (iNet ready) Panel to the iNet X Panel.
According to the App for the new Panel (https://play.google.com/store/apps/details?id=com.truma.inetx) all the Settings are changed via Bluetooth right now. So maybe its worth looking into the Bluetooth communication.

I am already using another ESP32 with ESPHome Software for various other things (I/O, Sensors) which is also using Bluetooth to get the state of the Gas Level from the Mopeka Sensors. So maybe that can be combined.

I dont understand the new Panel enough yet, but your software simulates an iNet Box correct? So normally the iNet Box would receive a command via SMS and then forward that command to the CP Plus Panel via the BUS protocol which would then again forward the command to the actual heater again using the BUS protocol? Is that correct?

nightcat91, 2023-01-02:
After some reading I think this page makes it a good summary: https://www.truma.com/de/de/faq/compatibility-inet-inetx
Again sorry to all the none German people for the German version.
The essential info is that the new  iNet X Panel is not compatible with the old iNet System. And it cannot be connected to an iNet Box for remote connections. Right now the new System can only be controlled via Bluetooth and Truma is still working on the new remote control feature!

That explains why the Panel was not reacting to the commands send by the "software iNet Box".

mc0110, 2023-01-02:
Yes, I'm so sorry, but this won't function

nightcat91, 2023-01-02:
Yeah sorry for maybe wasting your time with this while not checking compatibility first...

You mentioned a Telegram Group, could you point me in the right direction to that? Maybe something can still be done in this matter.
Maybe someone can reverse the Bluetooth communication between the mobile app and the Panel?
What would happen if I get an old CP Plus iNet ready Panel and connect that to the second RJ11 (LIN BUS Port) of my Truma heater. Could both panels control the heater? Could I then use the old CP Plus Panel to connect the inetbox2mqtt software to that? Would the new iNet X Panel update the status if the other panel changed settings?
Lots of questions....

I will also change the issue title to better reflect the actual issue!

closed 2023-01-03 by mc0110