mc0110/inetbox2mqtt #25: RX is working , but not able to TX from homeassistant....

amaral989, 2023-01-20:
Hello, 
I would like to start by saying , i appreciate all this hard work getting this to work!
I have been searching for a solution for truma communication and you have done it. I thank you!

im no engineer, but i have accomplish getting the RP2 loaded, and talking to HA. I'm able to see changes from truma when i change settings on the panel. But im not able to send any commands from my Home Assistant. 

Are you able to provide some guidance on where i should be looking. 

Thank you very much, 
John A

mc0110, 2023-01-20:
Hi John,

you mentioned:  _I'm able to see changes from truma when i change settings on the panel._
You would have to specify this. 

Do I understand correctly that you can make changes to CPplus and then you see the changed data in HA?
What exactly do you see there?
Which inetbox2mqtt version do you use?
Which HA setup are you using - is it the yaml from the docs?
Can you please use a mqtt monitor (e.g. mqtt explorer) and send one of the set-commands there.

Best regards
M.

amaral989, 2023-01-20:
Thank you for your response.

"Do I understand correctly that you can make changes to CPplus and then you
see the changed data in HA?"

   - Yes, I make changes to the CPplus control panel and it will show up on
HomeAssistant. For example, if I change from gas to electric, or if I
change the target heating temp on the panel. It will also show in HA.

Another example, if I want to change the fan on or off from HA. It will not
make changes to the control panel.

*Which inetbox2mqtt version do you use?*
I just started this project.. so i downloaded the latest version. Not sure
where to find the version number.

Which HA setup are you using - is it the yaml from the docs?
yes, im using your yaml docs.

Can you please use a mqtt monitor (e.g. mqtt explorer) and send one of the
set-commands there.
yes i can.. i will download it and gather as much info as possible.

I do have a few questions.... does LIN module have any other LED for
communication?, I only get the power led .
i do have few more modules on order.. to help troubleshoot. i have another
LIN module and a few more pico boards ordered.

thank you very much for helping,
John Amaral

***@***.***
908-463-9910

mc0110, 2023-01-20:
Hi John,

it seems to be a strange problem. 

In order for there to be communication between the simulated INET box and the CPplus, the init process must have taken place. This is a one-time process. But for this, the simulated INET box has to send some codes. This process must have worked for you, unless you already had a real INET box on your CPplus used before.
My thoughts are, that your rp2 could no longer transmit, that would explain the behaviour, because then it could receive codes, but could no longer transmit buffers.
The test scheme:
You can try it out by performing an init on the CPplus once, but disconnecting the rp2 beforehand. After that, there are only 2 entries in the init-setup. If you now reconnect the rp2 again and start the Init, it must log on again and for this it must send. If this does not happen, then this confirms the assumption that your transmission direction was interrupted after initialisation (TJA1020 or rp2 GPIO defective).

amaral989, 2023-01-20:
Ok. I will try your suggestion.
I did have a real inet box once installed a long time ago. But have removed
it many months ago.

So I can provide you with the most accurate info, you would like me to
disconnect rp2 and then reconnect it?
Also would like to let you know that I do have the LEDs hooked up to help
with visually see communication.

I do see the led flashing then goes steady on. Which I believe it's the
init process to the CPplus panel.

Amd the other led has a dim fast flashing when communicating to mqtt.

I believe I might have a hardware failure, as I tried two rp2 on a soldered
board.
I will soon try another rp2 and a new LIN module. It should arrive on
Sunday.

Thank you once again for your help.
It's the final step for my RV. I have integrated everything else using HA
and the truma has been the only piece of the puzzle.
Hopefully this will work out.

------ i do have a few esp32 ethernet/wifi boards. Would those work as
well.  The pin do not match your description.
Would be nice to be able to communicate over cable.

mc0110, 2023-01-20:
John,

would you explain this, pls
_------ i do have a few esp32 ethernet/wifi boards. Would those work as
well.  The pin do not match your description.
Would be nice to be able to communicate over cable._

amaral989, 2023-01-20:
Yes...
I personally like to use esp32 microcontroller with ethernet port instead
of wifi.

I have used these boards successfully on other projects.

I have enclosed a picture of the board.

amaral989, 2023-01-23:
Hello again,
I received a new TJA1020 today and a rp2. 
I was able to follow your github and when I installed it in the van. It does the same as the previous boards. 

I get info from truma and it populates in home assistant. 
But I'm not able to control the truma control panel. 

I had a mqtt explorer on a laptop on same network. and I'm able to see the messages. But truma will not react to the changes that I select in homeassitant. 

I tried to run the automation, which it does run and send info via mqtt. But it will not control the truma at all. 

Could it be something in the programming? This is a USA rv, with truma combi. 
I don't think it's the hardware.

I did at one point had inet hooked up and it worked without issues. 
If you can help me make this work, i will repay you for your time. 

Thank you so much! 
[image]
[image]

mc0110, 2023-01-23:
Hi John,

I am very sorry that you are having these problems.

At the moment I am not sure whether your problems have resulted from the changeover of Sw from Rel. 1.15.a to Rel. 2.0.1. Unfortunately, there was a bug in 2.0.1 that would explain your error. It would be really helpful if you would please install the latest version 2.0.2 once again. If this solves your problem, I apologise in advance for the inconvenience.

If the problem with the change persists, it is not due to the inetbox2mqtt software. 

Further steps for error analysis:
1) Version information of TRUMA, CPplus and INETBOX in the INIT menu.
2) Establish a terminal connection to the rp2 pico and record the LOGS (especially when sending a command). On windows the recommended sw is putty.
Please make sure that you do not reveal your credentials, they can be seen in the logs during the start-up process.

Finally, one question:
What do you mean by "I did at one point had inet hooked up and it worked without issues." Do you mean Daniel's inetbox version?

amaral989, 2023-01-23:
No need to apologize.
Actually should be thanking you for your help.

I have confidence this will eventually work out. I will work on this issue
this evening to gather the info you requested.

At one point , 2.0.1 was the version I was running. I will do another
upgrade to update it to 2.0.2
I agree with you that this doesn't seem like a issue with inet2mqtt.
It almost seems that the truma is not recognizing the commands being sent
to it.

I do have a question though... on the LIN module. It has a power led. But
does it have any other LED indicating communication?

In HA Lovelace, I see one thermostat. Is there suppose to be another
thermostat for water temperature control?

What I meant by inet is I had trumas actual INET box hooked up and I was
able to use trumas app through Bluetooth to control truma functions.

Thanks again
John Amaral

amaral989, 2023-01-24:
Updated to 2.0.2, and i still have same issue. 
Truma control panel will not accept the commands from homeassistant. 

i tried to gather all the software/firmware version... 
rp2 is running micropython 1.19.1
truma combi 4e has firmware version 5.7
CPplus firmware is 3.0

im also noticing that my sensor "alive" is set to off, and the strangest thing is the "target temp room payload" keeps climbing. 
starts at 32, then it jumps to 45, then 106, then 223 and continues 

sensor for truma time doesnt populate also. 

just trying to provide as much info as possible as i am completely lost. But have hope we can get this to work. 

thanks  

John

amaral989, 2023-01-24:
[image]

mc0110, 2023-01-24:
Hey John,

to be honest, it doesn't look good!

The microPython version 4358 (you can see it in your browser image) is OK, because version 1.19.1 has been around for more than half a year. But you seem to be using the latest version. In summary: micropython is OK!

Your version information from CPplus, on the other hand, looks strange. We see normally entries with

- C04.03.00 -> 0050.00
- H1.02.01 -> 0340.0b

in the Init menu for CPplus and Heater. I don't understand your information. Where did you get them from?

The ***alive sensor*** is very important. If the **alive sensor is off**, the TRUMA doesn't send any 18er-requests (CPplus asks the inetbox for messages). Accordingly, the inetbox cannot send any messages. This is a sure sign that the INIT process was not successful. 

Please show me your wiring between **LIN - TJA1020 - rp2**. Have you set the ground connection correctly?

My suggestion on how to proceed:
1. Switch CPplus off and on again. Pls disconnect the box and don't reconnect the inetbox2mqtt for this step.
1. Carry out INIT without inetbox2mqtt -> when the INIT has been carried out, there are only two entries in the init menu (C..... and H....).
1. Connect inetbox2mqtt to the CPplus, inetbox2mqtt running in "normal-run" mode (we hope that the connections are correct).
1. Carry out INIT again -> if the INIT was successful, there should be now 3 entries in the Init menu.
1. Now ALIVE should change to ON

From this point, the time of the CPplus should be transmitted every minute
.... and ALSO the commands, you send via the mqtt broker should control the TRUMA.

I keep my fingers crossed

amaral989, 2023-01-24:
Sorry.. 
Here is the correct info. 
Cpplus : C3.00.00
Heater: H5.07.01
Thermostat? : 23.70.0

[image]

[image]
[image]

mc0110, 2023-01-24:
Thanks for the data of the TRUMA / CPplus.

The 3rd entry T23.70.0 is the inetbox2mqtt. The entry is the same for me.

At the moment I assume that your heating is already compatible.

For clarification:
Let's have one further look at the INIT process. To do this you have to select and confirm the menu item RESET in the CPplus menu and then also confirm the PR SET that appears. The display then shows a flickering INIT...

Please send me the photos of the installation and try to follow my suggested plan.

amaral989, 2023-01-24:
Ok.
When I get home.. I'll send you pics and info on how I have it wired up..

Thank you

---- I live in the USA, would it be easier for you to teamview into my
laptop for additional troubleshooting.

---- Like I said, I'm willing to compensate your time and effort to make
this work.

Thank you again
John Amaral

amaral989, 2023-01-27:
I have followed your procedure on initializing the cpplus.  The index shows three firmware version. The keypad, heater, and inet emulator (rpi2)

I still get no clock or alive. 
I powercycled the heater, the cpplus, inet emulator numerous times. 
The alive never becomes on or active. 
I triple checked my wiring. 

Like I said before, if I make changes on the keypad. It will update on homeassistant. 

Could it be cpplus , maybe a later firmware will work?
Perhaps I can downgrade the firmware as re on the inet emulator?

I really wished this would work as it's my final price of my puzzle. 

I have enclosed some pics of the wiring.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mc0110, 2023-01-27:
It is very surprising that I now have two users at the same time who have an early CPplus version C3.00.0x and comparable difficulties. I therefore suspect that it is due to the CPplus version.

Since the microPython version of the LIN-module only uses a lightweight state-machine for the control, we can't get any further. 

It would therefore be helpful if you could test Daniel's raspi version [inetbox](https://github.com/danielfett/inetbox.py). If the CPplus works with Daniel's version (it has a different structure, so this is not unrealistic), please write a log file with ***DEBUG_LIN=1 truma_service** and test not only INIT but also the writing of commands.

amaral989, 2023-01-27:
Could it be as simple as purchasing a new cpplus unit, with perhaps a
recent firmware update?

If it's that simple.. I'm ok with that idea.
I'm willing to make this work with your system.

Thank you
John Amaral

amaral989, 2023-01-27:
Are you located in USA?
I can send you my cpplus panel if it would assist you in anyway!

mc0110, 2023-01-27:
New information - you need a CPplus Version C4.03.00. With C3.00.0x it won't be running. Sorry for the bad news! I live in Germany.

mc0110, 2023-01-27:
But I asked the expert, the new one is able to talk with your heater

amaral989, 2023-01-27:
This might be to much to ask, if I don't find one here. Is it possible you
can purchase one and ship it to me?

I will pay in advance!

USA truma market doesn't allow purchasing items. I would have to purchase
outside of the great USA 😔

mc0110, 2023-01-27:
https://www.ebay.de/itm/125736570472?mkcid=16&mkevt=1&mkrid=707-127634-2357-0&ssspo=ryAwbSo8QZm&sssrc=2349624&ssuid=Ez_yGSurTdW&var=&widget_ver=artemis&media=MORE

mc0110, 2023-01-27:
this means around $200 - it is cheap here and the seller means, never used

mc0110, 2023-01-27:
but there are more than 40$ for shipping - I think, this is very expensive

amaral989, 2023-01-27:
Perfect.. I'll order it asap.

Hopefully this will be the resolution..

I'm working on a project where everything is controlled by homeassitant..

I'm currently using homeassistant and esphome.

Thanks

mc0110, 2023-01-27:
Same for meVon meinem iPhone gesendet

mc0110, 2023-01-27:
But what is your definition of everythingVon meinem iPhone gesendet

amaral989, 2023-02-01:
Fyi... 
A new truma cpplus is on its way to my home. 
I placed an order with truma for a new controller. 

Hoping it will work afterwards 😆 🤣

mc0110, 2023-02-01:
I think, we will be successful! :-)Von meinem iPhone gesendet

amaral989, 2023-02-06:
Hello... 
Wanted to provide you with an update. 
I installed the new CPplus and after initlizing. I am now able to see the "alive", the time is updating . And I'm able to control truma. 

But I do want to ask you if you ran into a problem with the automation. 
The automation is constantly changing the temperature on thermostat, and turning on the water heater. 

I have enclosed a few pictures.
[image]
[image]
[image]
[image]
[image]

The target temperaute just multiplies...

mc0110, 2023-02-06:
Hi John,

first of all, I would say **Congratulations**!

Even though we haven't completely solved the problem yet, it sounds really good!

Let's approach it systematically:
I suspect it is because you are using Fahrenheit and not Celsius. 
It would be a good idea if you would first switch off the HA automations and we'll see if the control with MQTT commands works.
Would you please try that and let me know the result?
In the second step, if this works, we will try if you can see the HA entities (they all start with truma_) in HA and if the selects also work.

mc0110, 2023-02-06:
John, thanks a lot for your sponsorship!

amaral989, 2023-02-06:
It's my pleasure.. it's something very small for you help in creating this
and helping me out.

I'm no software engineer or programmer of any sort.
But this project made it simple for me to get this truma working with
homeassitant.

Have you worked with esphome? Would something like this work on esphome
rather then mqtt?
Thank you again
John A

amaral989, 2023-02-06:
Yes,  I'm using F , not C.
I have not reach my home yet to continue troubleshooting.
But last night, I disabled two of the automation. I believe it was
temperature change automation, and one more.

All the other automation  worked ok.

I noticed the water heater on/off automation worked but not the eco ,or
boost selection.

When I get home I'll do it again and take pictures.

Thanks
John

mc0110, 2023-04-06:
Hey John,

I hope you didn't just quit in frustration but got the solution to work. 

Is there anything else I can do for you?
Best
Magnus

closed 2023-05-18 by mc0110