mc0110/inetbox2mqtt #26: Using this code without HA - push button control of water heating

andypnz, 2023-01-25:
Hi Magnus!

Thank you for this great code repository!

I am trying to set up a push button (via RPi Pico W and LIN bus) to turn on the water heater (Truma Combi 6e) to 'HOT' + a LED, then once the water heater reaches temperature turn it off, and also turn off the LED. I dont have HA in my camper, and I would like to not need to connect to wifi/mqtt either.

I have managed to remove the HA requirements (I think!) but I really am not sure how to 'nicely' remove the wifi connection / mqtt requirements?

My main problem is currently the code runs (without needing HA, but connected to wifi), but when I press my button I get this error when I press my button:

`>>> %Run -c $EDITOR_CONTENT
INFO:connect:Detected micropython 1.19.1 4102 on port: rp2
INFO:__main__:Normal mode activated - for chance to OS-mode type in terminal:
>>>import os
>>>os.remove('run_mode.dat'
INFO:main1:Found Raspberry Pico Board, using UART1 for LIN on GPIO 4(tx), 5(rx)
INFO:main1:1
INFO:main1:Board LED test
INFO:main1:main-loop is running
INFO:connect:Try to open wifi and mqtt connection
INFO:connect:Credentials found: {'SSID': 'Banana', 'MQTT': '192.168.0.93', 'HOSTNAME': 'arpPico', 'ADC': '0', 'ASL': '0', 'WIFIPW': '65e00e9402', 'UN': 'mqtt_user', 'UPW': 'cressida1'}
INFO:connect:STA connection connected successful
{'sta_mac': b'(\xcd\xc1\x07\xb2\xde', 'sta_dns': '8.8.8.8', 'cred_fn': True, 'cred_bak': False, 'run_mode': 1, 'run_mqtt': False, 'sta_txpower': 31, 'port': 'rp2', 'ap_state': 'off', 'python': 'micropython 1.19.1 4102', 'sta_mask': '255.255.255.0', 'sta_ssid': 'Banana', 'sta_channel': 1, 'sta_ip': '192.168.0.46', 'sta_gateway': '192.168.0.1', 'sta_state': 'on'}
INFO:mqtt_async:Connecting to ('192.168.0.93', 1883) id=b'e6614864d3731b34' clean=1
INFO:mqtt_async:Connecting to ('192.168.0.93', 1883) id=b'e6614864d3731b34' clean=0
INFO:connect:Wifi connected: 192.168.0.46
INFO:connect:MQTT connected
INFO:main1:lin-loop is running
INFO:main1:hwc-loop is running
INFO:main1:Button pressed
INFO:main1:INFO:main1:error1
Task exception wasn't retrieved
future: <Task> coro= <generator object 'lin_loop' at 20024bb0>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "main1.py", line 243, in lin_loop
  File "lin.py", line 278, in loop_serial
AttributeError: 'bytes' object has no attribute 'hex'
INFO:main1:Received command: current_temp_room payload: 0
INFO:main1:inet-key:current_temp_room value: 0
INFO:main1:Received command: current_temp_water payload: 0
INFO:main1:inet-key:current_temp_water value: 0
Water heater was turned off 5 sec ago, so now lets set it to hot
INFO:main1:Water heater is now set to HOT
INFO:main1:Received command: energy_mix payload: gas
INFO:main1:inet-key:energy_mix value: gas
INFO:main1:Received command: target_temp_water payload: 0
INFO:main1:inet-key:target_temp_water value: 0
INFO:main1:Received command: operating_status payload: Off
INFO:main1:inet-key:operating_status value: Off
INFO:main1:Received command: alive payload: OFF
INFO:main1:inet-key:alive value: OFF
INFO:main1:Received command: heating_mode payload: off
INFO:main1:inet-key:heating_mode value: off
INFO:main1:Received command: el_power_level payload: 0
INFO:main1:inet-key:el_power_level value: 0
INFO:main1:Received command: target_temp_room payload: 0
INFO:main1:inet-key:target_temp_room value: 0`

Changes made to the code:
1. tools.py - added another LED to PIN_MAP
2. connect.py - added the LED to PIN_MAP, country changed to NZ, turned off my new LED
3. cred.py - returned q = "2.0.2" so it does not update
4. main1.py:
- removed the del and set ha_autoconfig(connect.client) commands
- added my button definition, etc
- added this after "async def lin_loop":
'async def hwc_loop(buttonReady, manualHWCboost):
    await asyncio.sleep(1)  # Delay at begin
    log.info("hwc-loop is running")
    while True:
      if (buttonReady == 1):
            if (button.value() == 0): # button pressed
                log.info("Button pressed")  
                try:
                    lin.app.set_status("target_temp_water", "0")
                    log.info("target_temp_water = ", "0") 
                    buttonReady = 0
                    set_led("redLED", False)
                    #redLED.on()     #(this turns it OFF)
                except:
                    log.info("error1")
                    # send via mqtt
                await asyncio.sleep(5)  # Wait 5 seconds
                print("Water heater was turned off 5 sec ago, so now lets set it to hot")
                try:
                    lin.app.set_status("target_temp_water", "60")
                    log.info("Water heater is now set to HOT")
                    manualHWCboost = 1
                    set_led("redLED", True)
                    #redLED.off()     #(this turns it ON)
                    await asyncio.sleep(60)  # Wait 60 sec before continuing          
                except:
                    log.info("error 2")
                    # send via mqtt
      if (manualHWCboost == 1):
            await asyncio.sleep(60)  # Wait 60 sec before continuing
            log.info("Current HWC Temp: ", lin.app.get_status("current_temp_water"))
            if (int(lin.app.get_status("current_temp_water")) > 59 ):
              log.info("HWC finished heating, turn off")
              try:
                lin.app.set_status("target_temp_water", "0")
                log.info("HWC now OFF, target_temp_water = ", "0") 
                manualHWCboost = 0
                buttonReady = 1
                set_led("redLED", False)
                #redLED.on()     #(this turns it OFF)
              except:
                log.info("error 3")
                # send via mqtt                              `

andypnz, 2023-01-25:
`async def hwc_loop(buttonReady, manualHWCboost):
    await asyncio.sleep(1)  # Delay at begin
    log.info("hwc-loop is running")
    while True:
      if (buttonReady == 1):
            if (button.value() == 0): # button pressed
                log.info("Button pressed")  
                try:
                    lin.app.set_status("target_temp_water", "0")
                    log.info("target_temp_water = ", "0") 
                    buttonReady = 0
                    set_led("redLED", False)
                    #redLED.on()     #(this turns it OFF)
                except:
                    log.info("error1")
                    # send via mqtt
                await asyncio.sleep(5)  # Wait 5 seconds
                print("Water heater was turned off 5 sec ago, so now lets set it to hot")
                try:
                    lin.app.set_status("target_temp_water", "60")
                    log.info("Water heater is now set to HOT")
                    manualHWCboost = 1
                    set_led("redLED", True)
                    #redLED.off()     #(this turns it ON)
                    await asyncio.sleep(60)  # Wait 60 sec before continuing          
                except:
                    log.info("error 2")
                    # send via mqtt
      if (manualHWCboost == 1):
            await asyncio.sleep(60)  # Wait 60 sec before continuing
            log.info("Current HWC Temp: ", lin.app.get_status("current_temp_water"))
            if (int(lin.app.get_status("current_temp_water")) > 59 ):
              log.info("HWC finished heating, turn off")
              try:
                lin.app.set_status("target_temp_water", "0")
                log.info("HWC now OFF, target_temp_water = ", "0") 
                manualHWCboost = 0
                buttonReady = 1
                set_led("redLED", False)
                #redLED.on()     #(this turns it OFF)
              except:
                log.info("error 3")
                # send via mqtt`

mc0110, 2023-01-25:
Hi Andy,

AttributeError: 'bytes' object has no attribute 'hex' indicates that the micropython version you are using does not work for us. There is also a notice about this in the README. You have to use one of the current nightly builds, please.

I have debugged your code and reduced it as desired and then tested it in my RV. You can also find the log file attached. Please put the file button_lin.py as main.py on the rp2, then the code will be executed at boot time.
[button_lin.zip](https://github.com/mc0110/inetbox2mqtt/files/10497734/button_lin.zip)
I keep my fingers crossed that you are successful.

If you have any other questions, don't hesitate.
If you are satisfied, I would be happy about a sponsorship. You will find the button on the project page.

andypnz, 2023-01-25:
Hi Magnus

Thank you! Ahh... I mis-read the documentation and searched for an earlier build of micropython. I will change to a nightly.

And thank you so much - really amazing - I did not expect you to do the code for me! 

I will test first thing tomorrow morning and report back

Thanks again!

mc0110, 2023-01-25:
Andy,

I thought it would be better if you could be sure that the code works with a TRUMA. So I also noticed that the time between switching off and then switching on the heater is 5 seconds too short. The only problem with testing was that it is winter here and only 4°C in the RV ;-)

But if I have understood correctly, you have not yet had any communication with your TRUMA. In this case, please also note the INIT note in the README. This also applies to your Sw.

andypnz, 2023-01-25:
Thanks again - I will be sure to perform the INIT process…. 

I have the new micropython + your software running on the bench - ready to try in the morning when there is some daylight :-)

andypnz, 2023-01-25:
Hi Magnus

Some testing this morning:

1. The registration process does not work when using the button_lin.py (as main.py)
2. Registration does work if I use the original main.py - so at least this shows the LIN bus board is working, and there is good communication between the Pico / Lin adaptor / CPplus
3. Using button_lin.py as main - the water heater does not turn on after button press. No errors in the log - so it looks like it should be working.... Any ideas what I can try to trouble shoot this?

Thank you - I think this is close to working :-)

andypnz, 2023-01-26:
Some small update.

If I add lin.loop_serial() to the hwc_loop in button_lin.py (as main.py) registration now works with this code.

So even though lin_loop and hwc_loop should both have been running (?), it looks like lin_loop was not?

Anyway, this [buuton_lin.py allows registration....](https://www.dropbox.com/t/GW42q5FGkmkQfqm3)

Unfortunately still no control - but the water temperature is probably correct ..... there is weak flasshing / flickering of the LED

INFO:__main__:Found Raspberry Pico Board, using UART1 for LIN on GPIO 4(tx), 5(rx)
INFO:__main__:1
INFO:__main__:Board LED test
INFO:__main__:lin-loop is running
INFO:__main__:hwc-loop is running
INFO:__main__:Button pressed
INFO:__main__:target_temp_water = 0
INFO:__main__:Water heater was turned off 5 sec ago, so now lets set it to hot
INFO:__main__:Water heater is now set to HOT
INFO:__main__:Current HWC Temp: 20.0
INFO:__main__:Current HWC Temp: 20.0

mc0110, 2023-01-27:
I don't understand your comments about the uasyncio. The LIN module and the control logic have different time frames. They cannot work together. Since my software example worked here in my RV, it was already clear to me that there must be other reasons for it not working for you. But never mind!

You have already mentioned the parallelism with John's issue! It is very surprising that I now have two users at the same time who have an early CPplus version C3.00.0x and comparable difficulties. I therefore suspect that it is due to the CPplus version.

Since the microPython version of the LIN-module only uses a lightweight state-machine for the control, we can't get any further. It would therefore be helpful if you could test Daniel's raspi version [inetbox](https://github.com/danielfett/inetbox.py). If the CPplus works with Daniel's version (it has a different structure, so this is not unrealistic), please write a log file with **DEBUG_LIN=1 truma_service** and test not only INIT but also the writing of commands.

mc0110, 2023-01-27:
New information - you need a CPplus Version C4.03.00. With C3.00.0x it won't be running. Sorry for the bad news!

andypnz, 2023-01-27:
Hu Magnus

Thank you - actually, that is good news! We could otherwise spend a long time going no where….

So you think heater firmware 5.00.00 is ok?

I will see what is possible to upgrade the CPplus :-)

Thanks again for all your help - I will let you know how I get on

mc0110, 2023-01-27:
My expert said that the new one (CPplus) can communicate with older heating systems, but not vice versa.

andypnz, 2023-01-27:
Great! Do you know if a newer CPplus will also work with inetbox2mqtt?

For example on eBay I see C4.04.01 - should that all work as well? They are expensive :-)

mc0110, 2023-01-27:
Yes, they will!

andypnz, 2023-01-27:
Great - thanks for all your help….

I will order one, and when it arrives let you know how it goes :-) Maybe this also solves the lin_loop / hwc_loop registration difficulty…..

andypnz, 2023-02-09:
Hi Magnus

My local RV dealer swapped my version 3.00.00 CPplus for a used 4.03.00 - which was great!

My first issue - I am trying to read the time from the CPplus.... 

               ` currentTimeString = lin.app.get_status("clock")`

and I get this error:

future: <Task> coro= <generator object 'hwc_loop' at 3ffeefe0>
Traceback (most recent call last):
  File "uasyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 178, in hwc_loop
  File "inetboxapp.py", line 433, in get_status
KeyError: clock

Any thoughts on that? 
Thank you!

mc0110, 2023-02-09:
Hi Andy,

first of all, congratulations that it now works for you too, if that was already clear after John's experience.

About your problems: 
The time is only transmitted once every minute. If you request it too early, you will get this error, because the key is only created after the time has been transmitted for the first time. The inetbox cannot force this transmission, it is provided by the CPplus.

The transfer protocol between inetbox and CPplus is a buffer transfer. Inetbox2mqtt looks at the individual elements and whenever (but only when) an element has changed, then the buffer with all elements is retransmitted. You can see the transfer in CPplus on the display, where an "APP" appears for 2-3 seconds. You should therefore make sure that you have not set the target_temp_water to "0" again somewhere.

andypnz, 2023-02-09:
Thank you, and thank you for your help!

Ok, I will ensure I wait > 60 seconds before asking `currentTimeString = lin.app.get_status("clock")`

Referring to your button_lin.py - would it be a good idea to do a 60 second time.sleep immediately before `e=asyncio.create_task(hwc_loop(buttonReady, manualHWCboost)) `

Rational: Lin_loop will already be running - so a 60 second sleep here should ensure coms are up and running and the key will be created

I have tested with your button_lin and I see this same behavior of water heater turning off after some 30 seconds or so. I will do some more testing tomorrow - maybe waiting longer after power on will help.... But I don't believe I am incorrectly setting the target_water_temp to "0" again.

andypnz, 2023-02-09:
Ok, I added a 60 second sleep after the lin_loop is started but before doing anything else - this fixed the clock issue :-)

And the water heater is working as expected - actually, it was yesterday - but I could not hear it so thought it was turning off (after the initial fan noise) - but letting it run I now see the water temp rising as expected :-)

Thank you for bringing to life a great project!

mc0110, 2023-02-09:
Thanks for your sponsorship

closed 2023-02-10 by andypnz