mc0110/inetbox2mqtt #28: Feature Request:  Serial to MQTT

Maik2016, 2023-02-07:
Hello, I like the box, I want to use the ESP32 to read out the victron MPPT controller, would it be possible to just make a serial -> one way only MPPT to MQTT gateway with 2 Serial ports (2x MPPT controller) ?  This would be very generic and also could use other hardware over serial MQTT readout

mc0110, 2023-02-07:
Hi,
the ESP32 has several hardware UARTs. This should therefore be possible. Since the UARTs also run via IRQs, there should also be no timing problems. If you look at the add-on examples (spiritlevel and duocontrol), all the essential parts are included. But since the ESP32 has only little free memory, you can't use all the add-ons anyway. Everyone has to find their own compromise. 

The concept is certainly generic, but each serial communication is its own problem. The topics / payloads used and the error handling are special and require their own routines. Here it certainly makes more sense to use ESPHOME, for example, which offers a [serial sensor](https://esphome.io/cookbook/uart_text_sensor.html).

Maik2016, 2023-02-07:
Yes i know, I programmed my own solution on an esp32 but it is to special and I having only on esp for truma and this it would be power efficient.  I have no idea about python but I'll look in the files you mentioned. 

I have also an ESP with 16MB should this be better in the long-run what did you think ?

closed 2023-04-06 by mc0110