mc0110/inetbox2mqtt #29: Change MQTT broker port via web Site

rjkrause, 2023-02-15:
Good day,
during testing, I ran into one problem, namely the impossibility of setting a different port for the MQTT broker. The default is set to 1883, but I use the service of the shared MQTT broker in the cloud and there the port is different (19340).
I do not have the possibility to install the environment so that I can make the modification myself in the source code, so I am asking if it would be possible to make a small modification in the WEB client so that the port can be set in Credentials.

thanks for support

RJK

mc0110, 2023-02-15:
Hi,
I understand your problem and regret very much that the possibility to choose the port does not exist at the moment. Allow me to point out, however, that I must strongly advise against controlling your heating via a public, unencrypted broker.

Unfortunately, the adaptation is not quite as simple as you make it out to be and requires the modification of several modules and, especially for the ESP32, the compilation of new firmware. I will take this into account for the next major release change.

As a workaround I have the following suggestion for you. Please add a line in the module main1.py in line 310.
This should make it possible to adjust the port, even though I have not tested this myself.

    connect.config.set_last_will("service/truma/control_status/alive", "OFF", retain=True, qos=0)  # last will is important
    connect.set_proc(subscript = callback, connect = conn_callback)

    connect.config.set_last_will("service/truma/control_status/alive", "OFF", retain=True, qos=0)  # last will is important
    connect.config.port = 19340
    connect.set_proc(subscript = callback, connect = conn_callback)

This should establish the connection. Let me know if it worked.

rjkrause, 2023-02-16:
**Good evening.**
Thanks for your help, I made the edit and the MQTT connection is working. However, I have another problem, apparently this is related to the fact that I do have a Truma Combi 4E heater in the caravan, but in the UK version and also with a CP25 control panel. It was probably not tested with this configuration.

I managed to turn on DEBUG mode in some strange way (I really don't know how, and I don't even know how to turn it off :), I see this in the output:

```
18:05:55.391 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
18:05:55.391 -> configsip: 0, SPIWP:0xee
18:05:55.391 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
18:05:55.391 -> mode:DIO, clock div:2
18:05:55.391 -> load:0x3fff0018,len:4
18:05:55.391 -> load:0x3fff001c,len:4252
18:05:55.391 -> load:0x40078000,len:11920
18:05:55.453 -> load:0x40080400,len:3344
18:05:55.453 -> entry 0x4008060c
18:06:01.806 -> INFO:connect:Detected micropython 1.19.1 10502 on port: esp32
18:06:02.443 -> INFO:__main__:Normal mode activated - for chance to OS-mode type in terminal:
18:06:02.443 -> >>>import os
18:06:02.443 -> >>>os.remove('run_mode.dat'
18:06:05.223 -> INFO:main1:LIN-LOG defined
18:06:05.223 -> INFO:main1:INET-LOG defined
18:06:05.223 -> INFO:main1:MQTT-LOG defined
18:06:05.223 -> INFO:main1:Found ESP32 Board, using UART2 for LIN on GPIO 16(rx), 17(tx)
18:06:05.223 -> INFO:main1:Activate duoControl set to true, using GPIO 18,19 as input, 22,23 as output
18:06:05.223 -> INFO:main1:main-loop is running
18:06:05.872 -> INFO:connect:Found Credentials
18:06:12.297 -> ...INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=1
18:06:17.229 -> INFO:main1:lin-loop is running
18:06:17.484 -> INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
18:06:18.289 -> INFO:connect:Wifi connected: 192.168.83.214
18:06:18.336 -> INFO:connect:MQTT connected
18:06:18.336 -> DEBUG:main1:Set subscription
18:06:25.200 -> DEBUG:main1:b'service/truma/set/target_temp_room' b'17'
18:06:25.200 -> INFO:main1:Received command: target_temp_room payload: 17
18:06:25.200 -> INFO:main1:inet-key:target_temp_room value: 17
18:06:25.200 -> DEBUG:inetboxapp:set_status: target_temp_room:17
18:06:36.383 -> INFO:main1:set ha_autoconfig
18:06:46.220 -> INFO:main1:duo_ctrl-loop is running
18:06:49.927 -> DEBUG:main1:publish energy_mix:gas
18:06:50.176 -> DEBUG:main1:publish el_power_level:0
18:06:50.377 -> DEBUG:main1:publish target_temp_water:0
18:06:50.579 -> DEBUG:main1:publish target_temp_room:17.0
18:06:50.781 -> DEBUG:main1:publish alive:OFF
18:06:50.983 -> DEBUG:main1:publish heating_mode:off
18:06:51.598 -> DEBUG:main1:publish operating_status:Off
18:06:51.791 -> DEBUG:main1:publish current_temp_room:0
18:06:52.045 -> DEBUG:main1:publish current_temp_water:0
18:06:52.245 -> DEBUG:main1:publish duo_ctrl_gas_red:OFF
18:06:52.445 -> DEBUG:main1:publish duo_ctrl_i:OFF
18:06:52.608 -> DEBUG:main1:publish duo_ctrl_ii:OFF
18:06:52.846 -> DEBUG:main1:publish duo_ctrl_gas_green:OFF
18:07:14.533 -> DEBUG:lin:in < 00 55 3c 01 04 b8 01 03 02 ff ff 3c
18:07:14.734 -> DEBUG:lin:in < 00 55 03 22 0b ff ff ff ff ff ff cf
18:07:14.815 -> DEBUG:lin:in < 00 55 c4 0e 0b ff ff ff ff ff ff 22
18:07:14.935 -> DEBUG:lin:in < 00 55 85 01 ff ff ff ff ff ff ff 79
18:07:15.016 -> DEBUG:lin:in < 00 55 06 00 00 ff ff ff ff ff ff f9
18:07:15.137 -> DEBUG:lin:in < 00 55 14 aa 0a 22 0b dc 0a 01 ff 22
18:07:15.218 -> DEBUG:lin:in < 00 55 55 00 00 ff ff ff ff ff ff aa
18:07:15.298 -> DEBUG:lin:in < 00 55 d6 00 07 34 0b 1d 0c 31 85 03
18:07:15.419 -> DEBUG:lin:in < 00 55 3c 01 06 b2 23 17 46 01 03 c1
18:07:15.621 -> DEBUG:lin:in < 00 55 3c 01 04 b8 01 03 02 ff ff 3c
18:07:15.822 -> DEBUG:lin:in < 00 55 03 22 0b ff ff ff ff ff ff cf
18:07:15.903 -> DEBUG:lin:in < 00 55 c4 0e 0b ff ff ff ff ff ff 22
18:07:16.024 -> DEBUG:lin:in < 00 55 85 01 ff ff ff ff ff ff ff 79
18:07:16.104 -> DEBUG:lin:in < 00 55 06 00 00 ff ff ff ff ff ff f9
18:07:16.225 -> DEBUG:lin:in < 00 55 14 aa 0a 22 0b dc 0a 01 ff 22
18:07:16.306 -> DEBUG:lin:in < 00 55 55 00 00 ff ff ff ff ff ff aa
18:07:16.427 -> DEBUG:lin:in < 00 55 d6 00 07 34 0b 1d 0c 3b 85 f8
18:07:16.507 -> DEBUG:lin:in < 00 55 3c 01 06 b2 23 17 46 01 03 c1
18:07:16.709 -> DEBUG:lin:in < 00 55 3c 01 04 b8 01 03 02 ff ff 3c
[... 48 lines cut ...]
18:07:22.712 -> DEBUG:lin:in < 00 55 06 00 00 ff ff ff ff ff ff f9
18:07:22.812 -> DEBUG:lin:in < 00 55 14 aa 0a 22 0b dc 0a 01 ff 22
18:07:22.913 -> DEBUG:lin:in < 00 55 55 00 00 ff ff ff ff ff ff aa
18:07:23.012 -> DEBUG:main1:publish alive:OFF
18:07:23.012 -> DEBUG:lin:in < 00 55 d6 00 07 34 0b 1d 0c 31 85 03
18:07:23.113 -> DEBUG:lin:in < 00 55 3c 01 06 b2 23 17 46 01 03 c1
18:07:23.314 -> DEBUG:lin:in < 00 55 3c 01 04 b8 01 03 02 ff ff 3c
18:07:23.513 -> DEBUG:lin:in < 00 55 03 22 0b ff ff ff ff ff ff cf
18:07:23.614 -> DEBUG:lin:in < 00 55 c4 0e 0b ff ff ff ff ff ff 22
18:07:23.714 -> DEBUG:lin:in < 00 55 85 01 ff ff ff ff ff ff ff 79
18:07:23.816 -> DEBUG:lin:in < 00 55 06 00 00 ff ff ff ff ff ff f9
18:07:23.915 -> DEBUG:lin:in < 00 55 14 aa 0a 22 0b dc 0a 01 ff 22
18:07:24.014 -> DEBUG:lin:in < 00 55 55 00 00 ff ff ff ff ff ff aa
```

If I disconnect the cable from the panel to the heater, the output generates this data, after disconnecting the cable from the panel, the data generation stops. From this I conclude that the LIN converter is connected correctly - at least it receives data from the panel. This type of panel can [... cut ...]
Do you think it makes sense to deal with the possibility of modifying the firmware so that the control is functional even for the UK version of the heater, or should I just buy a new control board and panel? I found this on Ebay:

[https://www.ebay.de/itm/295500713879?hash=item44cd370f97:g:Xu0AAOSwkcJj0KF-&amdata=enc%3AAQAHAAAAoFvOaW6RhZkrVp%2F0H1quettzIVLEMjlbPeJ1aIJj0SQCiNZLARgsuizfcdD5JmR0nh5GpsFDu66DpJFCsGGOvVi52elnkShIHHKFQ8c6WhecIu9oO%2Bla0oT%2Fm%2BTPaT7aJAE8NmAcSwQhxKMyiXoMq8Xev8MbcTl01KS28BgpSql40mcEi1nNCQsFDgUtkEIjxm1NSiJhgnOMtocEIH0k%2BS8%3D%7Ctkp%3ABk9SR8DnlbzLYQ](url)

Would it work with that??

Thank you and best regards
KRAUSE R.J.

rjkrause, 2023-02-16:
Good evening,
yes, the CP25 doesn't really send me anything. If I reset the panel by unplugging the power it shows me version 1.01 :)
In the end, I went the route of purchasing the entire assembly, panel and control board. The old one still gives me errors and the heating goes out. It's still cheaper than buying a whole new heater. Those UK versions behave strangely here in the EU - no one even lets them be repaired. I might be able to sell the CP25 as a spare part to another English caravan owner.

The panel board should arrive within a week, I'll replace them and try again.

Another question about control - how do I switch the board back to service mode, e.g. to change the WIFI connection? So far, I've been doing this by flashing the entire firmware to the ESP32 and initializing from scratch. But that's probably not a good procedure. I am using the latest version of 

`flash_esp32_inetbox2mqtt_v210_4M.bin`

Thank you for your patience and help

Jacob

mc0110, 2023-02-17:
Hi Jacob,
I can understand the decision to renew the control board. It also sounds promising when everything can be controlled.

As described in the readme, there are two ways to switch back to OS mode.

1) Option - with a terminal connection
Normal mode activated - for chance to OS-mode type in terminal:

        import os
        os.remove('run_mode.dat')

2) Option - with an MQTT connection
send topic 'service/truma/set/os_run' with a payload '1'

rjkrause, 2023-02-22:
Hello,
I don't know what to do anymore. As I announced, I bought a new heating control board including the panel. From a legal point of view, it seems to work, but only partially.
The initialization of the panel looked for unity, according to the status I can see that it also communicates with the heating unit, but only in one direction. I can change the values, but I don't get the current values back, so I can't really see what I changed.

To clarify. Modifying the code in main1.py did not work. I only connect to the MQTT broker when I make a change in the mqtt_async.py library on line 455 where I change the port to the required 19340.

[image]

.....INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=1
INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
INFO:connect:Wifi connected: 192.168.8.100
INFO:connect:MQTT connected
INFO:lin:B2 - identifier for NAD 03
INFO:lin:B2 - initializer for NAD 03   -----------------> start registration
INFO:lin:B0 - init finalized - send ackn ---------------> registration finalized
ets Jun  8 20bB}0x00008060c
INFO:connect:Detected micropython 1.19.1 10502 on port: esp32
INFO:__main__:Normal mode activated - for chance to OS-mode type in terminal:
>>>import os
>>>os.remove('run_mode.dat'
INFO:main1:Found ESP32 Board, using UART2 for LIN on GPIO 16(rx), 17(tx)
INFO:main1:main-loop is running
INFO:connect:Found Credentials
.....INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=1
INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
INFO:connect:Wifi connected: 192.168.8.100
INFO:connect:MQTT connected
INFO:main1:lin-loop is running
INFO:main1:Received j:inet-key:energy_mix value: gYVH <T]}, in c_subsc]YVH2"main1.py", line 120, in callback
NameError: name 'exception' isn't defined
INFO:connect:Wifi connection lost
INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
INFO:connect:Wifi connected: 192.168.8.100
INFO:connect:MQTT connected
INFO:main1:Received e=main1:inet-key:energy_mption wasn't1, in run_until_K5  File "main1.py", line 120, in callback
NameError: name 'exception' isn't defined
INFO:connect:Wifi connection lost
INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
INFO:connect:Wifi connected: 192.168.8.100
INFO:connect:MQTT connected
INFO:main1:Receiveas
INFO:main1:inetexception wpy", lipy", line 86, in c_subscripted
  File "main1.py", line 120, in callback
NameError: name 'exception' isn't defined
INFO:connect:Wifi connection lost
INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
INFO:connect:Wifi connected: 192.168.8.100
INFO:connect:MQTT connected
INFO:main1:Received :inet-key:energy_mWVH <5  Fpted
  File "main1.py", line 120, in callback
NameError: name 'exception' isn't defined
INFO:connect:Wifi connection lost
INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
INFO:connect:Wifi connected: 192.168.8.100
INFO:connect:MQTT connected
INFO:main1:Received command: taZenergy_mix valuretrieved
fu run_until_comne 86, in c],5  File "main1.py", line 120, in callback
NameError: name 'exception' isn't defined
INFO:connect:Wifi connection lost
INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
INFO:connect:Wifi connected: 192.168.8.100
INFO:connect:MQTT connected
INFO:main1:Received yload: gas
JjTask exceptiocore.py", lionnect.py", 86, in c_subscripted
  File "main1.py", line 120, in callback
NameError: name 'exception' isn't defined
INFO:connect:Wifi connection lost
INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
INFO:connect:Wifi connected: 192.168.8.100
INFO:connect:MQTT connected
INFO:main1:Received5INFO:main1:inetTask exceptio	, line 1,ine 86, in c_subscripted
  File "main1.py", line 120, in callback
NameError: name 'exception' isn't defined
INFO:connect:Wifi connection lost
INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
INFO:connect:Wifi connected: 192.168.8.100
INFO:connect:MQTT connected
INFO:main1:Received :inet-key:energy_mix value: gtrieved
fucomplete
  F File "main1.py", line 120, in callback
NameError: name 'exception' isn't defined
INFO:connect:Wifi connection lost
INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
INFO:connect:Wifi connected: 192.168.8.100
INFO:connect:MQTT connected
INFO:main1:Received  gas

Then there is an attempt to send the data to the mqtt server, but I see a lot of errors in the log, which makes the ASP32 disconnect from the WiFi and restart again.

Can I ask you to compile the image for the ESP32 board with the mentioned change in **self._c.port = 8883 if config.ssl_params else 19340** ??

Alternatively, advice on how to proceed properly if I wanted to use the source codes and upload the whole thing?? I'm not completely adept at programming in this environment - I'm programming a PLC,

I think the problem will probably be that I'm wrongly uploading the modified file to the board, after uploading and restarting it gives me memory errors.

thank you for the advice

[image]
[image]
[image]

rjkrause, 2023-02-22:
If I try to send data directly from the MQTT broker, the values on the panel change, otherwise I have zero values. This information is displayed on the TrumaCP panel if I have an ESP32 connected.

[image]

[image]

Interestingly, only this information is sent to the MQTT broker

[image]

mc0110, 2023-02-22:
Hi Jacob,

    .....INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=1
    INFO:mqtt_async:Connecting to ('108.129.17.116', 19340) id=b'b4e62d932b51' clean=0
    INFO:connect:Wifi connected: 192.168.8.100
    INFO:connect:MQTT connected
    INFO:lin:B2 - identifier for NAD 03
    INFO:lin:B2 - initializer for NAD 03 -----------------> start registration
    INFO:lin:B0 - init finalized - send ackn ---------------> registration finalized

Let's briefly summarise what you can see in the logs:

Your LIN connection is up and running, the INIT process has been completed successfully. This means that the ESP32 is able to control your TRUMA.

Your MQTT connection is not working properly. The connection to the broker is established, but it does not work properly. Something is wrong. The Wifi connection keeps breaking, the transmitted payloads, but also some of the topics are not correct. As mentioned above, I recommend looking for another mqtt broker-solution.

mc0110, 2023-02-22:
`To clarify. Modifying the code in main1.py did not work. I only connect to the MQTT broker when I make a change in the mqtt_async.py library on line 455 where I change the port to the required 19340.`

This solution corresponds to the fixed programming of a port in the library - of course it works. 

If you have flashed the ESP32 with the modified firmware, I am only surprised that the change has an effect on your system. You must be using the ESP32 with our own micropython firmware, right?

closed 2023-04-06 by mc0110