mc0110/inetbox2mqtt #30: Change/Edit the Topic Path

jimslim2, 2023-02-19:
Hi Magnus,

thank you for your work!

Actually i am using the original iNet-Box, but i've been working on a solution, thats more flexible and more integrated.
More flexible in terms of compatibility to other components and more integrated in terms of usability and ui.
To accomplish this, i've installed a Victron Cerbo GX as kind of central point. The platform of the underlying Venus OS is quite open. Next to different kind of sensors, etc. i can connect, there is a mqtt broker on board, which i can use local, without being connected to the internet. This is a must have for me, because i have to keep an eye on the WAF ;-)
Furthermore the local broker forwards his messages to the VRM Cloud, wich is the Victron cloud portal.
Now i make use of it, by connecting to this central mqtt broker i.a. to get connected to my Home Assistant instance.

To get to the point:
This lokal mqtt broker of the Cerbo GX device needs a kind of prefix topic for incoming messages to be excepted, for instance: '_N/49e7da53012f/system/0/..._', where '_49e7da53012f_' is the ID of the installation.

So a topic of '_service/truma/control_status/target_temp_room_' won't be accepted by the broker.

My goal is to send a topic like this:
'_N/49e7da53012f/system/0/service/truma/control_status/target_temp_room_'

Would you please advice me how to get this done?

Thanks in advance
Daniel

mc0110, 2023-02-19:
Hi Daniel,

the installation doesn't sound easy. My WAF would fail without an internet connection to accompany it. But anyway:

You have to adjust main1.py line 46-49, but you will lose the auto-config function in HA, because it does not react if the topics look like you need it to. Can't you alternatively use the HA-mosquitto-instance?

    # Change the following configs to suit your environment
    S_TOPIC_1       = 'service/truma/set/'
    S_TOPIC_2       = 'homeassistant/status'
    Pub_Prefix      = 'service/truma/control_status/'

jimslim2, 2023-02-20:
Hi Magnus,

thank you for your quick response.
The HA mosquitto broker is not in the same network, so if i would go this way, i had to deal with vpn, etc.

So i will try to edit the lines from main1.py.

What is your solution? What kind of mqtt broker do you use in your RV?

mc0110, 2023-04-06:
Is there anything else I can contribute to this?

closed 2023-07-05 by mc0110