mc0110/inetbox2mqtt #38: Truma switches automatically on after HA restart

CzarofAK, 2023-03-28:
My HA is switching on the Room Heater after rebooting. I have also troubles to keep the truma switched off. Somehow it starts automatically!

CzarofAK, 2023-03-28:
[image]

mc0110, 2023-03-28:
I need more info about our setup! The HA automation is correct function only with Celsius degrees

CzarofAK, 2023-03-28:
Yes, of course, everything you need.
i have copied your truma.ii and yaml.
the only thing changed is, the min temp for a test (to switch it off via target room temperature setting.
i can switch it off on the truma panel.

mc0110, 2023-03-29:
Sorry, I don't understand the problem. 

If I understand correctly, you are using my ESP module and can also control the Truma via MQTT. The HA scripts have been tested many times and run perfectly in many installations. If you have changed something, you have probably made a mistake.

CzarofAK, 2023-03-29:
I use your code straight out the box. I have made changes to investigate the fault.

It seems like that the home assistant cant either send the data correctly to the esp or the esp cant send the data to the bus.

mc0110, 2023-03-29:
I don't understand what you mean?

mc0110, 2023-03-29:
The best thing to do is to reinstall ESP and HA Yaml.

closed 2023-05-18 by mc0110