mc0110/inetbox2mqtt #45: Error message with RP 2

rzaenger, 2023-07-11:
Hi Magnus, I just pulled your latest changes to my RP 2 and noticed that there is an error message coming out of the new web_os_main.py: 

[image]

With the previous version this did not occur, had seen that you have extended the code...

What could be behind this?

BR Roland

mc0110, 2023-07-12:
Hi Roland,

sorry for the problems the version change is causing you. 

Since there are now so many different boards with LIN decoders, each with its own pinout, the automatic recognition (RP2 <-> ESP32) has given way to a configuration dict. The name of the configuration must be stored as an entry in ***arg.dat***. Currently, the default entry is ESP32, but since most users flash the ESP32 as a BIN file, I will use RP2 as the default entry here. 
The entry in arg.dat must therefore be: **hw=RP2**

rzaenger, 2023-07-12:
Thank you Magnus,
Have created tha arg.dat and now it looks good:
[image]

Now i will going to check this in the campervan ...

mc0110, 2023-07-12:
:-)

rzaenger, 2023-08-17:
Hallo Magnus,

vielen Dank.

Gibt es auch andere Wege, als über Github und die Kreditkarte, Dir eine Spende zukommen zu lassen?

Beste Grüße
Roland Zänger

mc0110, 2023-08-17:
Es liest sich so, als wäre es erfolgreich gewesen!

rzaenger, 2023-08-18:
Hallo Marcus,

im Grunde nicht…ich bin auf dem Weg nach Norwegen und werde dort, wenn ich etwas Zeit habe und es dich interessiert, die Situation zusammenfassen und noch eine Idee für die Erweiterung formulieren?

Beste Grüße

R. Zänger

mc0110, 2023-08-18:
At the moment I'm lost. 
I don't understand, what you want to say? Are you speaking about add ons? And which situation is to summarise.

rzaenger, 2023-08-22:
Hi Magnus, I am sorry for confusion...will write in English now.
This particular topic is closed for me. I use a Raspi and try to be up to date with my system and your changes. I do not use a HomeAssistant installation in my CamperVan but I try to transfer the MQTT data via a Teltonika Rut241 router to bring this to my HA system at home. No succes until now. Question: I found a Wireguard implementation for an ESP32 in the internet, could you imagen to implement this in your system? Would be helpful a lot...probably not only for me... (https://github.com/ciniml/WireGuard-ESP32-Arduino) I can not asses if it is possible, have no programming experience. Have a good day. BR Roland

closed 2023-08-23 by mc0110