mc0110/inetbox2mqtt #47: P5.02.00

vincentfuhler, 2023-08-25:
Test setup 1:

Firmware from the last release v201_4M
ESP 32 debug board
Truma Vario Heat
Truma CP Plus -> Software according to menu: P5.02.00 (I have checked this multiple times and initialized the CP without other devices because I was expecting a value with C4XXXX)

Initialization works. CP Plus shows three or, when the e-kit is connected, four devices.

After the setup, Home Assistant shows that an MQTT connection exists.
When I change values, they do not make their way into Home Assistant. In the MQTT log, I first see that the initialization was successful, then a data packet, then the error "Error Key 4" is displayed.

When I send a command in Home Assistant, the CP display shows "APP". However, the command is not executed.

Test setup 2:

With the current firmware v264_4M, the problem unfortunately worsens. The initialization continues to run smoothly. MQTT connection exists. But now when I send a command, no "APP" appears on the screen anymore, and I am also not shown any incoming connections. In the MQTT log, I first see that the initialization was successful, then no further entries.

I suspect it is due to P5.02.00, have you ever heard/seen this? I can't find anything about it on the Internet.

mc0110, 2023-08-25:
Hi Vincent,

The CPplus can control different devices. These have different buffer codes. The sim software can only process already known device codes (currently Combi 4 or Combi6 and Aventa). The control commands for your heating are simply different, just as inetbox2mqtt cannot control the Alde heating. The fact that the "APP" no longer appears in the new version is due to the changed transmission logic. This has been optimised. 

I regret that I cannot give you any better news.

vincentfuhler, 2023-08-25:
Hello Magnus, darn, it looks like I have some work ahead of me.
I'm currently considering how best to proceed. I'm a developer myself. How would you go about adding a new device?

So far, I have put the device into LOG mode and am also receiving data:

`12:50:30.320 -> INFO:connect:Set RUN-Mode: 1
12:50:41.450 -> INFO:connect:Set RUN-Mode: 0 = OS-RUN
12:51:08.099 -> INFO:web_os:Soft reset chip
12:51:08.099 -> MPY: soft reboot
12:51:14.336 -> INFO:args:args_file found -> loaded
12:51:14.671 -> INFO:args:file: args.dat content: hw=ESP32 lin=debug
12:51:14.671 -> INFO:__main__:release no: 2.6.4
12:51:14.710 -> HW:  ESP32
12:51:15.010 -> INFO:connect:Detected micropython 1.20.0 10502 on port: esp32
12:51:16.187 -> INFO:__main__:OS mode activated
12:51:16.187 -> INFO:connect:set ap to: 1
12:51:17.213 -> INFO:connect:AP enabled: 192.168.4.1
12:51:18.510 -> INFO:connect:DHCP: True
12:51:18.510 -> INFO:connect:fixedIP: peter132!
12:51:18.510 -> INFO:connect:LAN: False
12:51:20.435 -> ...INFO:connect:Wifi connected: 10.0.1.202
12:51:26.074 -> INFO:connect:Found Credentials
12:51:26.074 -> INFO:connect:Start mqtt connect task
12:51:27.392 -> INFO:web_os_main:run lin:True inet:False webos:False naw:False file:False
12:51:27.392 -> INFO:web_os_main:run uart:2 tx:17 rx:16
12:51:27.392 -> INFO:web_os_main:HW-Check ESP32 esp32
12:51:27.392 -> INFO:web_os:init
12:51:29.710 -> INFO:web_os_main:Start nanoweb server
12:51:29.710 -> INFO:web_os_main:Start OS command loop
12:51:29.749 -> INFO:mqtt_async2:Connecting to ('10.0.0.2', 1883) id=b'7821849dff5c' clean=1
12:51:30.484 -> INFO:mqtt_async2:Connecting to ('10.0.0.2', 1883) id=b'7821849dff5c' clean=0
12:51:30.773 -> INFO:connect:Interface connected: 10.0.1.202
12:51:30.773 -> INFO:connect:MQTT connected
12:51:31.060 -> INFO:web_os_main:lin-loop is running
12:52:19.259 -> DEBUG:lin:_B2 - response request
12:52:19.316 -> DEBUG:lin:out > 03 06 f2 17 46 00 1f 00 87
12:52:22.553 -> INFO:lin:B2 - identifier for NAD 03
12:52:22.628 -> DEBUG:lin:out > 03 06 f2 17 46 00 1f 00 87
12:52:23.664 -> INFO:lin:B2 - initializer for NAD 03   -----------------> start registration
12:52:23.711 -> DEBUG:lin:out > 03 06 f2 17 46 00 1f 00 87
12:52:24.787 -> INFO:lin:B0 - init finalized - send ackn ---------------> registration finalized
12:52:24.833 -> DEBUG:lin:out > 03 01 f0 ff ff ff ff ff 0b
12:52:26.960 -> DEBUG:lin:out > fe ff ff ff ff ff ff ff 28
12:52:27.008 -> DEBUG:lin:_B2 - response request
12:52:27.031 -> DEBUG:lin:out > 03 06 f2 17 46 00 1f 00 87
12:52:29.678 -> DEBUG:lin:out > fe ff ff ff ff ff ff ff 28
12:52:29.734 -> DEBUG:lin:_B2 - response request
12:52:29.795 -> DEBUG:lin:out > 03 06 f2 17 46 00 1f 00 87
12:52:30.305 -> DEBUG:lin:out > fe ff ff ff ff ff ff ff 28
12:52:30.375 -> DEBUG:lin:_B2 - response request
12:52:30.413 -> DEBUG:lin:out > 03 06 f2 17 46 00 1f 00 87
12:52:30.933 -> DEBUG:lin:out > fe ff ff ff ff ff ff ff 28
12:52:31.005 -> DEBUG:lin:_B2 - response request
12:52:31.041 -> DEBUG:lin:out > 03 06 f2 17 46 00 1f 00 87
12:52:31.552 -> DEBUG:lin:out > fe ff ff ff ff ff ff ff 28
12:52:31.626 -> DEBUG:lin:_B2 - response request
12:52:31.664 -> DEBUG:lin:out > 03 06 f2 17 46 00 1f 00 87
12:52:32.173 -> DEBUG:lin:out > fe ff ff ff ff ff ff ff 28
12:52:32.243 -> DEBUG:lin:_B2 - response request
12:52:32.276 -> DEBUG:lin:out > 03 06 f2 17 46 00 1f 00 87
12:52:32.830 -> DEBUG:lin:out > fe ff ff ff ff ff ff ff 28
12:52:32.899 -> DEBUG:lin:_B2 - response request
`

PS: Is there a way to turn off the access point mode?

Pre10der, 2024-06-17:
@vincentfuhler - did you make any progress with support for the Truma VarioHeat gas heaters by chance? I am looking for a solution for my caravan to be able to control the heater via a mobile device.