mc0110/inetbox2mqtt #51: MQTT with TLS ?

el-Presi, 2023-09-29:
Hi there,

well as this a project, which should take place in a battery operated environment, I would like to avoid having a rpi in my camper. I try to send MQTT via the internet directly to my home-homeassistant. To do this, every normal guy would use MQTT on port 8883 (TLS) - It seems (? I cannot see any logs) that a TLS MQTT connection is not possible?

mc0110, 2023-09-29:
Hi, 
what do you mean with "normal guy". TLS is not tested at the moment, but it doesn't work with the library either. Otherwise, I am amazed at the new constellations that I am always asked about.

el-Presi, 2023-09-29:
Hi ! Sorry for the issue here :)
TLS is working here on my side (I guess it does - I had no wireshark in between yet, but the broker should only accept tls connections on 8883 port - and this workes :) So I was wrong here! Sorry!!

"Normal guy"-part: lets say it otherwise: "normally you want to have an encrypted connection, when talking over the internet"

closed 2023-09-29 by mc0110