mc0110/inetbox2mqtt #52: ESP32 no (re-)connect

Stiles96, 2023-10-07:
I flashed an ESP32 and the system is running.

By starting (turn on enrgy in my RV), the FritzBox ist starting also and the ESP32 don't connect to the wifi because the ESP is fast than the wifi of the fritzbox. I have to reset the ESP by pressing the reset-button.

I have also the reconnect issue after a few hours. The system lost the wifi connection and didn't connect anymore until I reset the ESP.

mc0110, 2023-10-07:
What exactly is the issue? My solution works if a WiFi connection is available when booting. This does not seem to be the case in your system. I have tested a reconnect after a lost WiFi connection countless times and it works. There must be something else wrong with you.

Stiles96, 2023-10-07:
[image]

In the screenshot I have the time line from the truma_clock. At 5:03 there was a disconnect and then no reconnect. It can also be a issue by the reconnection of the ISP. Every 24 hours the systems get a new IP address.
I think, the mqtt session is interuppted.

At the boot time there is no wifi running. Fix: No wifi connection ==> reboot after 2 minutes to reconnect or try a reconnect. 

In my other ESP32 project I used this code for wifi setup und reconnect. It works very good:
ˋˋˋ
WiFi.mode(WIFI_STA);
  WiFiMulti.addAP(SSID, wifipassword);
  WiFi.onEvent(WiFiStationDisconnected, SYSTEM_EVENT_STA_DISCONNECTED);  //Call Function, if Wifi Disconnect.
  esp_task_wdt_init(WDT_TIMEOUT, true); //enable panic so ESP32 restarts
  esp_task_wdt_add(NULL); //add current thread to WDT watch
  // wait for WiFi connection
  Serial.print("Waiting for WiFi to connect...");
  while ((WiFiMulti.run() != WL_CONNECTED)) {
    Serial.print(".");
  }
  esp_task_wdt_reset();
  Serial.println(" connected");
  ˋˋˋ

Stiles96, 2023-10-07:
Is there a log available to check?

mc0110, 2023-10-07:
There are debug functions in the software. These are controlled via the file args.dat. If you do not use a monitor (e.g. Thonny), you can also load the file via the file manager in OS mode up/down.
In the file, please add the following in addition to the hw=ESP32:

hw=ESP32 connect=debug

Stiles96, 2023-10-08:
I checked the log. After ISP reconnect the mqrr broker gets a new IP Address. The ESP32 try to reconnect, but the system uses the old ip address and make no DNS Lookup until reboot.
The mqtt disconnect triggers a wifi reconnect.

Hint: in the log there is the wifi password listed as fixedIP.

INFO:main1:HW-Check ESP32 esp32
INFO:main1:prefix: 'truma' set: service/truma/set/ rec: service/truma/control_status/
INFO:mqtt_async2:Connecting to ('80.141.32.241', 1883) id=b'94b97ed312d4' clean=1
INFO:mqtt_async2:Connecting to ('80.141.32.241', 1883) id=b'94b97ed312d4' clean=0
INFO:connect:Interface connected: 192.168.181.16
INFO:connect:MQTT connected
INFO:main1:lin-loop is running
INFO:main1:del ha_autoconfig completed
INFO:main1:set ha_autoconfig completed
INFO:main1:Initializing completed
INFO:main1:inet-key:aircon_operating_mode value: off
INFO:main1:inet-key:target_temp_water value: 0
INFO:main1:inet-key:heating_mode value: off
INFO:main1:inet-key:aircon_operating_mode value: vent
INFO:main1:inet-key:aircon_vent_mode value: mid
INFO:main1:inet-key:aircon_operating_mode value: vent
INFO:main1:inet-key:aircon_vent_mode value: mid
INFO:main1:inet-key:aircon_operating_mode value: vent
INFO:main1:inet-key:aircon_vent_mode value: mid
INFO:main1:inet-key:aircon_operating_mode value: vent
INFO:main1:inet-key:aircon_vent_mode value: mid
INFO:main1:inet-key:aircon_operating_mode value: vent
INFO:main1:inet-key:aircon_vent_mode value: mid
INFO:connect:Wifi connection lost
INFO:connect:DHCP: True
INFO:connect:fixedIP: ***Wifi password***
INFO:connect:LAN: False
..INFO:connect:Wifi connected: 192.168.181.16
INFO:mqtt_async2:Connecting to ('80.141.32.241', 1883) id=b'94b97ed312d4' clean=0
WARN:mqtt_async2:error in MQTT reconnect.[Errno 113] ECONNABORTED
INFO:mqtt_async2:Connecting to ('80.141.32.241', 1883) id=b'94b97ed312d4' clean=0
WARN:mqtt_async2:error in MQTT reconnect.[Errno 113] ECONNABORTED
INFO:mqtt_async2:Connecting to ('80.141.32.241', 1883) id=b'94b97ed312d4' clean=0
[same to messages continues until reboot]

closed 2023-11-06 by mc0110