mc0110/inetbox2mqtt #53: Ventilation Mode

Stiles96, 2023-10-07:
System:
ESP32, Truma CP Plus, Truma Combi 4

At the Truma CP Plus Panel I can select the Ventilation Mode. To turn it on, I have to select mode "vent" and can choose the speed by 1 - 10.
It's the same ventilation, wich is use by the heating mode. By heating the ventilation only have eco and fast.

The mqtt has some aircon values to set the system to vent, but not the speed. I don't know, if it works.

I also have some pictures to show that on the CP Plus:
 
[image]
[image]
[image]

mc0110, 2023-10-07:
Sorry, You couldn't control the truma vent mode with my solution. If you use a aventa aircon, it is possible to set the aircon vent-mode

mc0110, 2023-10-07:
This isn't a bug

closed 2023-10-07 by mc0110