mc0110/inetbox2mqtt #54: Using Micropython 1.2.0

gedger, 2023-10-11:
I'm trying to get this working on a Wemos LOLIN S3 mini. https://www.wemos.cc/en/latest/s3/s3_mini.html#

Unfortunately it was released after your recommended build of micropython. The first version available is 1.19.1 which doesn't seem very stable. However, the device works well with V1.20.0 and I've managed to load all the code onto the device and it connects to my MQTT server fine. I've had to change the tools.py file to use  UART0 together with the correct pins. This all works fine until I connect it to my Truma Combi (confirmed compatible). I've attached a log file. I'm not sure whether these issues are caused by the micropython version or by the differences in the ESP32S3. 

Are there any tests I can do to confirm the compatibility of micropython 1.20.0 or suggestions on how to debug.

Thanks

Richard

[ESPs3 Connection Logs.txt](https://github.com/mc0110/inetbox2mqtt/files/12867677/ESPs3.Connection.Logs.txt)

mc0110, 2023-10-11:
I have already had experience with the S3, but not with the Wemos design. The chip only has 2 UARTs. In principle, the software works, but not the full flash prints as is possible with the ESP32. The problem is that part of the python code must already be in the firmware, otherwise there is not enough memory. In addition, micropython does not currently support memory paging. That's why the bigger chips with more memory don't make sense. Try a normal ESP32 and if you like a smaller design, take an RPI pico and you will be successful.

mc0110, 2023-10-11:
Here is an S3 version for testing. After flashing, you must download all modules that are not already loaded on the chip under 'help modules()'.
[flash_pekaway_inetbox2mqtt_v251_2M.bin.zip](https://github.com/mc0110/inetbox2mqtt/files/12868524/flash_pekaway_inetbox2mqtt_v251_2M.bin.zip)
Allow me to point out that I do not support the version

gedger, 2023-10-11:
Many thanks, I will give that a try and report back.

gedger, 2023-10-11:
To start from scratch...

```
esptool.py erase_flash
esptool.py write_flash 0 flash_pekaway_inetbox2mqtt_v251_2M.bin 
```

Fails with

```
esptool.py v4.5.1
Found 2 serial ports
Serial port /dev/cu.usbmodem14201
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:a8:d7:74
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Unexpected chip id in image. Expected 9 but value was 5. Is this image for a different chip model?

A fatal error occurred: flash_pekaway_inetbox2mqtt_v251_2M.bin is not an ESP32-S3 image. Use --force to flash anyway.
```

gedger, 2023-10-12:
I've moved over to using the esphome component which works fine with my device so I'll close this request.

Thanks

Richard

closed 2023-10-12 by gedger