mc0110/inetbox2mqtt #55: MemoryError: memory allocation failed, allocating 776 bytes

lembnau, 2023-10-25:
Hi,

nach ein paar Stunden bekomme ich im seriellen Monitor (Putty) diesen Fehler und es findet keine Kommunication mehr statt

mc0110, 2023-10-25:
which device?
which mode (os-mode or standard)?
what do you mean with "no communication" -> http, mqtt, lin

mc0110, 2023-10-25:
Normally, when you are in http-communication, you see this more often (unfortunately, the chips don't have enough RAM and micropython unfortunately packs the "normal" code into the RAM area

mc0110, 2023-10-25:
The error is not really a software error. A memory error is triggered when you try to fetch the subscription. This is not possible in the versions that is in the repo. Without a complete log and details about the version and also the hardware you are using, I can’t help.

Carstens-Paulette, 2024-05-02:
IT seams that I have the same issue.
I will try to get some detailed logs about it.
In case, the ESP is simply running out of memory, how to deal with it? One approach would be to switch off the controller by the MQTT receiver. Im my case the VenusOS, which fortunatly have relays. Or is it possible to reset the ESP on low memory?

Follow up to be expected.

kind regards

Carsten O.

mc0110, 2024-05-02:
Hi Carsten,

that is very unlikely. Inetbox2mqtt is now very often used and has been tested for many months. Are you using the LAN connection and the WOMOLIN hardware? I really need the logs and the point at which the ESP32 exits. I myself run the code on an RPI, which has a larger RAM area. Jan has used the newer ESP32s, but the memory is switched via a mapping, which the micropython does not support.

closed 2024-06-03 by mc0110