mc0110/inetbox2mqtt #56: Is HA the only way to use this?

NicZ1, 2023-11-06:
Hello, sorry if I am being naïve but I am trying to understand how this works. 
I have bought an ESP32-Wroom and flashed it with your code.

At boot I get an hotspot and if I connect from a computer then I get the web frontend where I can set my credentials.  I setup the SSID and password of my internet router and then I switch it to run in normal mode.

When I reboot the device, I can see on my router that it is connected and if I ping the IP it replies happily.

Now what do I do? If I open a browser and try to connect to the IP I get "connection refused" so I am wondering if Home Assistant is the only way to interface this.

Is it really not possible to have a Web frontend in normal mode even if it is very simple? I was hoping to replace the Truma inet box as I don't like the way it interface with SMS messages. I was thinking that this would give me a easy web based interface that I can use to turn the heating on / off, set the temperature and check its status.

If I need to setup HA then I would need another box where to install the HA server (rather than pay an online subscription) but this will require extra power. So I was hoping that I could do everything form one little box. 

Many thanks for your assistance,
Nic.

mc0110, 2023-11-06:
So somehow I don't know exactly what your question is. In fact, many people use this solution without HA, because there is also an ESPHOME variant for HA.

NicZ1, 2023-11-06:
Many thanks for your reply. 
I thought that ESPHome was an add-on to Home Assistance but maybe I am missing something. I am starting to look into this but haven't found anything yet. 
Do you set it up with HA first then use it standalone? Do you have any links/pointers on how I could do this? Can I access this ESPHome setup on the internet while I am away from the local network?

mc0110, 2023-11-06:
It was never the project idea to build a web service. You are welcome to get involved.

closed 2023-11-06 by mc0110