mc0110/inetbox2mqtt #57: Alde 3030 with two separate heating circuits

justselling, 2023-11-06:
Hi, 
Is there already experience with an Alde 3030 with two separate heating circuits? In such a case, the room temperatures exist twice.

mc0110, 2023-11-06:
Hi,
sorry for the bad news: Alde commands aren't included.

justselling, 2023-11-06:
Oh, what a mess. I thought those two were compatible at the LIN bus level. So, you would first need to capture the commands from the Alde device on the bus itself using a debugger.

mc0110, 2023-11-06:
You are right! On the LIN bus level the logic is the same, but they have different command-buffers. At the moment I have no clear debug and on the other hand it needs an Alde test system. Unfortunately I do not have that

justselling, 2023-11-06:
I will try to remove my bench next week to gain access to the Alde. After that, I can connect the TI-CI-LIN controller. Then, I should be able to use the USB port to debug. Perhaps you can assist me with that?

marcvl64, 2023-12-07:
I have an ALDE 320 system (US model) and I'm very interested in figuring out how to remote control it from Home Assistant, since my motorhome is pretty much standardized on Home Assistant for everything. If I can help you figure out what needs to be done, I'd be glad to help.

mc0110, 2024-02-11:
Hi, it is not planned to integrate the communication with ALDE heaters in this software, because it is almost impossible to debug the code if you do not have the heating system available. 
However, as far as I know there is already a very good reverse engineering of the ALDE code. I would therefore refer you to the very good implementation of the LIN protocol by Fabian, who may already have implemented ALDE sequences. `https://github.com/Fabian-Schmidt/esphome-truma_inetbox `

mc0110, 2024-02-11:
I hope the information is helpful

closed 2024-08-16 by mc0110