mc0110/inetbox2mqtt #58: request: run ob Raspberry Pi 3B or similar

pkoevesdi, 2024-02-04:
I got the project https://github.com/danielfett/inetbox.py to run on my Raspberry Pi almost out of the box. From this project here I took the home assistant integrations - lovelace card and truma.yml.
Now, what I'm missing are some entities, probably because the danielfett-project doesn't decode them, such as truma clock, and in general I cannot see any values inside home assistant, I can only set some of them.
Now, I wonder, how can I get these projects merged, in case the functionality of the project of Daniel Fett is not a strict subset of this propject here (beside it's ability to run on linux), or how can I get this project here to run on a full linux system with the LIN connected to a serial port?.

mc0110, 2024-02-04:
Try to use the right hardware, this will be successful

pkoevesdi, 2024-02-05:
I have the right hardware for my project. Now I try to get the right software. This project here seems more mature, but Daniel Fett's project runs on Linux. I want to contribute a best-of-both version and try to figure out which is the best way. But seems, here is no interest in contributions?
Anyway, I figured out, mainly the descovery messages are missing at Daniel Fett, so it seems the shorter route to improve over there.