mc0110/inetbox2mqtt #60: Problems connecting to the mqtt broker

Deleny74, 2024-02-11:
Hello, 
I'm having problems connecting between my ESP and the Mosquitto Broker. My hardware is a Raspberry Pi4 with Homeassistant and an AZDelivery ESP32 Dev Kit C V4. For the instalationI used the esphome web tool. After installation I have access the board via webOS. The connection to the Truma heater is also working. I can send the two commands to heat the water. This works well. After restarting, the ESP is connected to my WiFi.

I made the other entries in the file editor according to the instructions. But, I don't get any data from the ESP. In the LOG File is an unknown IP Adress with a request but not with my IP address is 192.168.178.1.

2024-02-11 19:56:20: New connection from 172.30.32.2:55820 on port 1883.
2024-02-11 19:56:20: Client <unknown> closed its connection.

I installed mqtt explorer. Here I only see the topics that I send from the home assistant.

I do not know how to continue. I also tried a different ESP module and an older version of inetbox2mqtt. always without success.

In desperation I installed a d1 mini with the image for the Victron solar controller. Here I immediately get a connection with the mqtt broker.

Does anyone else have a tip for?

Thank you.

Greetings Andreas

mc0110, 2024-02-11:
So after your first paragraph, you have already achieved a lot. 

You can test the MQTT connection in OS mode, but you don't have a connection to the Home Assistant server there. You only get this when you change the operating mode. Which LOG files are you referring to? Obviously you are not talking about the ESP logs. You can also see logs on the ESP via a terminal connection. That would probably help.

Deleny74, 2024-02-12:
Hi, please can you explain how I can test the MQTT connection in OS mode?  I have allready changed in the Credentials the Network SSID and Wifi passcode to my Home Net. Also I have edit the Broker User & password. The Hostname ist the IP adress from my Homeassistant. I have stored the Creds and made a soft reboot (without change to run mode). I found the ESP in my NEtwork an I can connect the webOS with the new IP Adresse.

I have connect the Lin uart Adapter with the ESP and the Truma. If I press "Water Heater on" the Display from my Truma print App an after that the Water Heater works. Is that what you mean for MQTT connection test? 

The Log file what I mean ist the "Protokoll" from mosquiotto broker.  

2024-02-11 19:56:20: New connection from 172.30.32.2:55820 on port 1883.
2024-02-11 19:56:20: Client closed its connection.

here ist the LOG from my ESP32: there is one entry I was wondering about it.   

INFO:connect:fixedIP: J***1. thats my Wifi password (I have changed some Letters)

ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4672
load:0x40078000,len:14536
load:0x40080400,len:3364
entry 0x400805cc
INFO:args:args_file found -> loaded
INFO:args:file: args.dat content: hw=ESP32
INFO:__main__:release no: 2.6.5
HW:  ESP32
INFO:connect:Detected micropython 1.20.0 10502 on port: esp32
INFO:__main__:OS mode activated
INFO:connect:set ap to: 1
INFO:connect:AP enabled: 192.168.4.1
INFO:connect:DHCP: True
INFO:connect:fixedIP: J***1..
INFO:connect:LAN: False
...INFO:connect:Wifi connected: 192.168.178.44
INFO:connect:Found Credentials
INFO:connect:Start mqtt connect task
INFO:web_os_main:run lin:False inet:False webos:False naw:False file:False
INFO:web_os_main:run uart:2 tx:17 rx:16
INFO:web_os_main:HW-Check ESP32 esp32
INFO:web_os:init
INFO:web_os_main:Start nanoweb server
INFO:web_os_main:Start OS command loop
INFO:mqtt_async2:Connecting to ('0.0.0.0', 1883) id=b'a842e3cd6744' clean=1
INFO:web_os_main:lin-loop is running
Task exception wasn't retrieved
future: <Task> coro= <generator object 'connect' at 3fff9dd0>
Traceback (most recent call last):
  File "asyncio/core.py", line 1, in run_until_complete
  File "mqtt_async2.py", line 494, in connect
  File "mqtt_async2.py", line 197, in connect
  File "mqtt_async2.py", line 252, in _as_write
  File "asyncio/stream.py", line 1, in write
OSError: [Errno 113] ECONNABORTED
Traceback (most recent call last):
  File "main.py", line 91, in <module>
  File "/lib/web_os_main.py", line 93, in run
  File "asyncio/core.py", line 1, in run_forever
  File "asyncio/core.py", line 1, in run_until_complete
KeyboardInterrupt:
MicroPython v1.20.0-283-g7d66ae603-dirty on 2023-08-21; ESP32 module with ESP32
Type "help()" for more information.
>>>

mc0110, 2024-02-12:
Hi,

the line
INFO:mqtt_async2:Connecting to ('0.0.0.0', 1883) id=b'a842e3cd6744' clean=1
shows the problem! You have **set** the address of your MQTT broker incorrectly in your credentials.

What you should see are the two lines:
INFO:mqtt_async2:Connecting to ('192.168...20', 1883) id=b'782184dec058' clean=1
INFO:mqtt_async2:Connecting to ('192.168...20', 1883) id=b'782184dec058' clean=0
(I do not show the IP in full)

Only when the second line (clean=0) is showing, the registration process with the broker was correct.

Deleny74, 2024-02-12:
Thank you for your support. I have allways writen the IP Adress from my Broker in the mask for "Hostname" in the webOS. Now I have writen the IP adress in the mask "Brokername/IP and the MQTT connection is working fine.

closed 2024-02-12 by Deleny74