mc0110/inetbox2mqtt #62: Hot Water boost after ESP32 reboot

Deleny74, 2024-02-26:
Hi, 

I have a problem after reboot ESP32. The Water heating starts allways the boost mode without any reasons. Is there configuration in the settings alfter reboot the ESP32? I have loose lots of Propane after I have made a reboot :-(
 
A other Problem is that my ESP32 loose the conection to my Homeassistant after an few hours. In that chase the Status of the Truma is on "Läuft" but the Truma Time is not actuelle anymore. I can reach the IP with an Ping from my Network. But no communication with the mqtt-broker.

Do you a possible reason for that?

Thanks for your support.

mc0110, 2024-02-27:
I would like to split it up as you are reporting two different errors.

1) The communication problem sounds a bit like the one described in this issue [https://github.com/mc0110/inetbox2mqtt/issues/52](esp32 no reconnect). I am pretty sure that it is due to the router configuration, which is necessary for the MQTT connection. If the MQTT connection is interrupted, the "last will" is activated so that the status in HA is changing to "off". If the ESP loses the connection to the INETBOX, it reboots automatically after 2-3 min. This error has no longer been reported since I made these adjustments in version 2.6.5.

2) In the provided version of the HA setup there are extra buttons for switching the TRUMA functions on/off. Does the activation also happen when HA is not connected? If the ESP reconnects to the broker, the last command could possibly be accepted there again. This would explain the behaviour.

Deleny74, 2024-02-28:
Hi,
thank you for your answer.  
For the first error I have changed the network setting to fix IP address. Now I have to wait. But I think that must be the correct solution.

For the second error I have tried to restart the ESP32 without the Homeassitant. In this chase the water boost does not start. After restart the HA I have no connection to the mqtt broker. Is this correct? I thought that the esp32 would connect to the HA after every reboot? I wait 10 minutes after reboot, not connected. After another reboot the connection to mqtt is working. 

I have also found the settings in HA witch has startet the water boost after reboot. I my chase I changed the settings many time with out any effect.  The switch changed after a while back to on. I am not sure how it works. I used the switch "Helfer" on an of an dthen the swich from the water boost changed again to on an the heating goes to 16 Grad. I switched both of but it changed his status back to an. Only after switching off the temperature on the CP Plus does the temperature remain off. 

I am to stuiped!

Deleny74, 2024-03-01:
Hi again,

for the first error, loosing the conection ESP32 to my Homeassistant. The connection was runnuing pretty good till today in the morning. Now is the Truma Clock not working an the Status switched to "not running". 
The issue [https://github.com/https://github.com/mc0110/inetbox2mqtt/issues/52](esp32 no reconnect) is not working in my chase. I have the same IP Adress in my Network and I can send and receive an Ping to the ESP.

for the Problem with the water booster, I have found in HA the automation Truma water set mode3. Since I have deaktivated this automation I haven't longer the problem. So long...

Deleny74, 2024-03-05:
Hi, 
has nobody the same problems? 

After reboot of my ESP32 the System is running for an few hours an the Status switched to "not running" and the clock stops. 

The Problem with the water boost is also alive. I cant find any reason for this reaction from the automation. Yesterday canged the heating for my water to 200°C without any reason. I wasnt at home and the temperature switched to 200°C. I have seen this in the Log for this automation.
"TRUMA Bidrectional Change Water Temperature" started from "truma_target_temp_water"  at 22:18 . I wasnt at home at this time an I changed

closed 2024-03-17 by Deleny74