mc0110/inetbox2mqtt #64: Two input boolean have the same name in your truma.yaml file

Karatzie, 2024-03-29:
Hello Dr. Christ.

Thank you for your great implementation of a Truma heating to a mqtt server.
This implementation offers a lot of possibilities to create real heating automation in Home Assistant.

I have a real Truma Inetbox on my RV but this device has not all fuctions what I am excpect from this device.

I integrate your inetbox2mqtt in my RV Home Assistant.

I check your truma.yaml file and found two input boolean with the same name.

input_boolean:
  truma_ctrl:
    name: "TRUMA Zentralschalter"
  truma_button:
    name: "TRUMA Heater State"
  truma_water_button:
    **name: "TRUMA Water State"**
  truma_water_autooff:
    **name: "TRUMA Water State"**
    
    I changed the auto off function for two water states.
    I add an auto off for eco.
    I searched the name and took the wrong truma_water_button and not the truma_water_autooff button.
    
    BR
    
    Andreas (karatzie)

mc0110, 2024-03-29:
Hi Andreas,

I'm glad you like the software. Thanks for the tip. The fact that the input_booleans have the same names is wrong. The boolean "truma_water_button" is not needed. What exactly have you changed? What has changed in the control system?

Karatzie, 2024-03-31:
Hi,

I have changed notting but I will also delete the not used button.

BR

closed 2024-03-31 by Karatzie

mc0110, 2024-03-31:
The button was superfluous, but also without function. It has already been deleted in the source code and I tested it today in the RV. The control works perfectly.