mc0110/inetbox2mqtt #65: Automations on your truma.yaml slow down the Home Assistant host

Karatzie, 2024-03-29:
Hello Dr. Christ.

My Raspberry PI3 was slow down with the automations loaded from your truma.yaml file.
Thats happend durig opening of your automations in Home Assistant to investigate the function.
With every opening of another automation, the system was a little bit slower.
I must reboot my HA host.

At the end, I migrated all your automations in the Home Assistant system and deleted your automations from your truma.yaml file.
My Raspberry PI3 runs now as fast as before.

This is an info and not realy a bug.
It is possible that HA has here a bug.

BR
Andreas (kartzie)

mc0110, 2024-03-29:
Hi,

In my HA installation in my RV, the Truma and Aventa automations have been running for more than 9 months without any problems. My hardware is an ODROID system, my INETBOX simulation runs on RPI. I think you still have a bug in your implementation. Nevertheless, thanks for the hint about the useless Boolean switch "truma_water_button" , which I have removed from the code. 

I assume that it is fine for you to close the two issues, you opened.

Karatzie, 2024-03-31:
Hi.
Your program works fine as expected.
My system will be slower if I open your automations in Home Assistant to investigate the Automation.
It seems, that Home Assistant itselfe have a problem to display the automatisation details.

BR
Andreas

closed 2024-03-31 by Karatzie