mc0110/inetbox2mqtt #67: Truma Aventa Compact plus

aloisbrau, 2024-04-18:
Hallo. Schreibe auf Deutsch, da ich nicht perfekt in Englisch bin. Könnte es aber auch mit Mühe in Englisch formulieren!
Die Anbindung über mqtt in Home Assistant funktioniert bei der Truma Heizung sehr gut.
Leider habe ich keine Truma Aventa Comfort sondern die Aventa Compact plus.
Diese wird leider nicht angesteuert.
Könnte es sein dass hierfür andere ID in der Aventa.yaml notwendig sind?
Vielleicht gibt es eine Möglichkeit über NodeRed?
Finde hierzu leider nichts, möchte aber auch bei mqtt bleiben!

Danke
Alois

mc0110, 2024-04-18:
Hallo Alois,
freut mich, dass die Truma geht!
Wenn die Aventa Compact plus nicht auf die Codes reagiert, dann ist das nicht auf der YAML-Ebene änderbar. Es tut mir dann leid, aber da gibt es keine Chancen einer Steuerung.
Das ist unabhängig von Node-Red / MQTT, hierbei geht es um die LIN-Kommunikationsebene.
Beste Grüße
Magnus

For all the english speaking members:
I'm pleased that the Truma is working!
If the Aventa Compact plus does not respond to the codes, then this cannot be changed at the YAML level. I'm sorry then, but there's no chance.
This is independent of Node-Red / MQTT, this is about the LIN communication level.

aloisbrau, 2024-04-24:
Hallo! Nach eingehenden Tests und Auslesen von Daten aus unterschiedlichen Konstellationen bin ich einen Schritt weiter.
Mit esphome truma-inetbox von Fabian Schmidt kommen zumindest Daten der inetbox von der Aventa Compact Plus.
Passt alles zwar noch nicht, aber wenigstens habe ich jetzt eine Datenverbindung über Lin-Bus.
Dort hat jemand auch eine Saphir Klima eingebunden. Scheint doch möglich zu sein. Ich bleibe dran!!!!

closed 2024-06-03 by mc0110

mc0110, 2026-01-09:
Hi, danke für die schnelle Antwort und die Versionsstände. Die Änderung des GPIO von 14 auf 25 sollte funktionieren. Ich bin mir nur nicht sicher, ob die interne LED "frei" ist, also wenn Du die so ansprechen kannst, dann geht es auch über die Software.

Bytheway: Bei mir im Auto habe ich auch einen PICO seit 2,5 Jahren im Einsatz. 
Gute Fahrt wünsche ich Euch!!