mc0110/inetbox2mqtt #68: switching to OS mode without reason

Carstens-Paulette, 2024-04-26:
At first, many, many for your time and your effort in this project.

I have the problem, that the WomoLIN box is loosing the WiFi connection to router, althrough it is approx. 10cm away. After this it is rebooting in "OS mode"

So I have two questions:
a) Does the check box "LAN support" also works on the WomoLIN-box. Doesn't seam to work for me.

b) Will the box restart in "Normal mode" after repowering? In this case I can program a relay to interrupt the power supply via VRM in case this happens, while I'm not in the camper.

Many thanks for your time and best wishes

Carsten

https://paulette.die-oswalds.net/umbauten/inet-box-alternative

Carstens-Paulette, 2024-04-26:
Addendum: Release is 2.6.4

mc0110, 2024-04-26:
Hi Carsten,

I'm pleased that you like the software. I developed the LAN configuration especially for Jan's board, which he provided me with for this purpose. However, you definitely need to change the version so that the connection errors can be rectified. In the end, I had a very stable connection. Please make sure that you specify the correct hardware, see the instructions. When rebooting, the box starts in "Normal mode". Only if there is no MQTT connectivity to the broker over 10 reboots does the fallback to OS mode take place. In most cases, the credentials were then set incorrectly.

I like your article and I'm very pleased that there is also an Openhab version. If you would like to publish the connection, you are welcome to do so via the repo. If you have any other questions, please don't hesitate to contact me.

Magnus

Carstens-Paulette, 2024-04-27:
Hello Magnus,

I would like to update to the release 2.6.5 , but I have the WomoLIN hardware.
I'ved tried it with the [flash_esp32_inetbox2mqtt_v265_4M.bin.zip] and I can connect to the WLAN and the MQTT broker, but not to the TiN-interface.
The [flash_womolin_inetbox2mqtt_v264_4M.bin.zip] is reporting it's release 2.64, as predictable.
Do you have a 2.6.5 version for WomoLIN hardware?
I hope it's not to hard to update this branch. 

In the moment, I spepped back to WomoLIN 2.6.4 this works, als long as thr router and the VenusOS don't get a restart.

Many thanks in advance.

Carsten

mc0110, 2024-04-27:
Hello Carsten,

sorry that you are running into problems.

1) Option:
If you want to use a LAN connection, V2.6.4 should be sufficient.

2) Option:
The images are predefined for a special HW configuration. This can be changed as a parameter. In order for the LIN interface to be recognised correctly, the hardware must be changed to the WOMOLIN interface assignment. After burning, you will find a file arg.dat in the file directory, in which the setting must be changed to hw=WOMOLIN. I didn't really understand why Jan chose this configuration.

Good luck
Magnus

Carstens-Paulette, 2024-04-27:
Hello Magnus,

that was easy! I though, i have to compile the whole project. I will document it on my website.
Are you interested in the Node-RED setup?

[image]

Many, many thanks!

Carstens-Paulette, 2024-04-27:
Works!

closed 2024-04-27 by Carstens-Paulette

mc0110, 2024-04-27:
Great - have fun