mc0110/inetbox2mqtt #70: WiFi Password in fixed IP field

Carstens-Paulette, 2024-05-04:
Hello Magnus,

while setting up the debug for my lost connection issue, I found another annoying thing.

The fixedIP-Field is filled with the WiFi-password, althrough the system is not set to use it.

[image]

Annotation: serial speed for WOMOLIN is 115.200 bps, just if anyondy is searching for it.

mc0110, 2024-05-04:
Hi,

relax and let me explain it again. The system has been tested many times and works. I have tested the WoMoLin board here for weeks and if you set the HW setting in the args.dat correctly, the LAN connection will also work. You have a small error (the PW is printed in the DEBUG), but the IP is set correctly.

Enjoy and have fun

Carstens-Paulette, 2024-05-04:
Hello Magnus,
sorry for getting me wrong.
The software is really good. 
If you already know about this, and it does not have any influence on the function, everything is okay.
I don't want to annoy to.

Kind regards

Carsten

closed 2024-05-04 by Carstens-Paulette

mc0110, 2024-05-04:
Most of the problems reported here are connection problems because, for example

- the broker is on the Internet
- a VPN connection is being used
- the MQTT server does not show a clean protocol
- encrypted connections (TLS) are to be used

None of this is covered by the software. My aim was to realise a solution that connects a broker located in the RV with the TRUMA.