mc0110/inetbox2mqtt #72: MQTT Client doesn’t reconnect

Besc0029, 2024-05-19:
Hi,
I just finished the project and was really happy to get everything started. But the night my router (including the MQTT-Broker) in the campervan automatically restarted and after that I didn’t get any MQTT Message. I think that the MQTT Client looses connection when the broker restarts and can’t reconnect.
Any idea how to fix it? The esp connects to the WLAN/Router and is still connected to the Truma Panel.
Thanx
Bernhard

closed 2024-06-03 by mc0110

Besc0029, 2024-06-03:
Sorry for the question: why is the issue completed? Is it fixed or is there no fix for the problem ?
Thanx

mc0110, 2024-06-03:
The problem could neither be clearly described nor reproduced. In my case, the systems connect inThe problem could neither be clearly described nor reproduced. In my case, the systems connect autonomously.

Besc0029, 2024-06-03:
After a reboot of the broker?