mc0110/inetbox2mqtt #75: Warning in Home Assistant

7xy, 2024-08-14:
Deprecated option used in mqtt publish action call
This stops working in version 2025.2.0. Please address before upgrading.
Deprecated payload_template option used in MQTT publish action call to topic service/truma/set/target_temp_room from payload template 5. Use the payload option instead. In automations templates are supported natively. Update the automation or script to use the payload option instead and restart Home Assistant to fix this issue.

7xy, 2024-08-16:
Found it out myself!

Just change the term "payload_template" to payload inside the file /homeassistant/packages/truma.yaml using the File Editor.

Then the repair is done!

mc0110, 2024-08-16:
Hi Hannes,

great job, thanks for figuring that out. I'll adjust the yaml at short notice.

7xy, 2024-08-18:
There also has to be done a change in the automations.yaml file! Just set all "payload_template" to just "payload"!