mc0110/inetbox2mqtt #76: Improve install instructions for OTA-Installation with mip

iBobik, 2024-08-15:
Hello,
I tried to install it on the current MicroPython v1.23.0 on the ESP32, but it does not work.
<details><summary>Details</summary>
<pre><code>
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drINFO:__main__:release no: 2.6.5
HW:  RP2
INFO:connect:Detected micropython 1.23.0. 11014 on port: esp32
INFO:__main__:OS mode activated
INFO:connect:set ap to: 1
INFO:connect:AP enabled: 192.168.4.1
INFO:web_os_main:run lin:False inet:False webos:False naw:False file:False
INFO:web_os_main:run uart:1 tx:4 rx:5
INFO:web_os_main:HW-Check RP2 esp32
INFO:web_os:init
INFO:web_os_main:Start nanoweb server
INFO:web_os_main:Start OS command loop
Task exception wasn't retrieved
future: <Task> coro= <generator object 'run' at 3ffd7f60>
Traceback (most recent call last):
  File "asyncio/core.py", line 1, in run_until_complete
  File "/lib/nanoweb.py", line 236, in run
  File "asyncio/stream.py", line 1, in start_server
OSError: -203
INFO:web_os_main:lin-loop is running
</code></pre>
</details> 

So I had to read really carefully and investigated what version you require:
> However, the software did not run with the latest stable kernel from July 2022

> My tests were done with upython-version > 19.1-608.

So which version? Checking availeble one:
> v1.20.0 (2023-04-26) .bin / [.elf] / [.map] / [Release notes]
> v1.19.1 (2022-06-18) .bin / [.elf] / [.map] / [Release notes]
> v1.18 (2022-01-17) .bin / [.elf] / [.map] / [Release notes]

„from July“ means v1.20.0 will not work but „> 19“ means v1.20.0 is first supported.

So trying all 3 versions.

v1.19.1 and v1.18:
```
>>> import mip
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'mip'
```

v1.20.0:
```
…
Downloading github:mc0110/inetbox2mqtt/src/update.py to /
Copying: //update.py
Done
update.py True
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4656
load:0x40078000,len:13284
ho 0 tail 12 room 4
load:0x40080400,len:3712
entry 0x4008064c
INFO:__main__:release no: 2.6.5
HW:  RP2
INFO:connect:Detected micropython 1.20.0 10502 on port: esp32
INFO:__main__:OS mode activated
INFO:connect:set ap to: 1
INFO:connect:AP enabled: 192.168.4.1
Traceback (most recent call last):
  File "main.py", line 90, in <module>
  File "/lib/web_os_main.py", line 14, in <module>
  File "lin.py", line 18, in <module>
MemoryError: memory allocation failed, allocating 136 bytes
MicroPython v1.20.0 on 2023-04-26; ESP32 module with ESP32
Type "help()" for more information.
```

The same happened on when restarted by power off.

My HW is [M5Stack Station](https://docs.m5stack.com/en/core/station_485) with ESP32-D0WDQ6-V3 240MHz dual core, 600 DMIPS, 520KB SRAM and Flash 16MB. Flashed with ESP32_GENERIC MP variant.

mc0110, 2024-08-15:
I can't believe it. Maybe it's badly described or you haven't read it properly. The ESP32 does not have enough memory. You can't install the software, you have to use the images I have prepared very intensively. I was very happy to do this, but the same questions after 1.5 years are exhausting. No, you don't have to donate anything, just install the software on the ESP and it will work.

iBobik, 2024-08-15:
Thank you for quick reply. No offence, but maybe it is badly described when there are repeating questions. ;-)

I had some issues with your prebuild image, so I tried different approaches. Why not when it is officially offered in the README. It least I can help with some fixes or docs improvements when I know how to build it and your opinions about it.

mc0110, 2024-08-16:
I don't understand exactly what issues you have with the images. But if you want to run the software on the ESP32, you have to compile the mpython files together with mpython. Only then will there be enough memory

mc0110, 2025-04-01:
What is the status here? Can the issue be closed?

closed 2025-04-06 by mc0110