mc0110/inetbox2mqtt #77: Request for Interface Compatibility with Alde 3020

AWcgn, 2024-10-13:
Hi,

If i connect to the ALDE 3020 TIN the communication is defect.
In the Alde Panel (Paneel: s737R300A514 EU) in the Point iNetBox i see flapping: 
- First: Truma iNetBox v23.70.0
- after 1-2 seconds: Not found (Nicht gefunden)
- then Truma iNetBox ????
- again: Truma iNetBox v23.70.0
next turn.

Is the wish of this project to control Alde?
What can be the reason or can i test? 
I'm new here, so if this is the false platform to address this please tell me and erase this.

best regards 
Andreas

janmarkuswilhelm, 2024-11-30:
Hi, 
I do have the same setup. 
Alde 3020 HE
TI-CI-LIN Controller v2.1

The MQTT and the Womolin Controller are working like they should after flashing.
 
I did not yet connect the Alde Control panel as my caravan is already in its winter location. So I just want to prepare everything to be able to operate this or debug this once I am there. 

So my question ist, if there is already an explenation or solutions for the described issue?

DeBuhr-dev, 2025-01-17:
Hello, 
I use an ALDE 3030. I was hoping that this would also be supported, since INETBOX supports it. Are there any plans to expand this? I find the project very exciting and would be happy to provide support with information.

mc0110, 2025-01-25:
Sorry, but there are no plans from my side