mc0110/inetbox2mqtt #82: Connection to Raspberry Pi as Hotspot not possible

schnico87, 2024-12-09:
Hi,

I'm sure that my problem isn't caused by this project, but maybe someone has an idea for a solution. I use a Raspberry Pi as a WiFi hotspot in my camper van (the signal is extended into the interior of the camper van using an external antenna). During the inetbox2mqtt installation routine, the connection works without any problems and I can install the packages. When I try to connect using the credentials, no connection is established. If I select a different WiFi network, the connection works. Does anyone have an idea for a solution?

mc0110, 2024-12-10:
This sounds like a credential problem. Please run through the initialization process again.

schnico87, 2024-12-11:
> This sounds like a credential problem. Please run through the initialization process again.

Thanks for your answer. I checked the cerdentials several times. I spend now some hours for this step. :( Unfortunatly for me it is not clear in the log wich step missed Login to Wifi or the connection to MQTT Broker. I tried to enter only the WiFi Credentials for the well working Connection i get the same messages. I attached the log from tera term to this post.

[image]

schnico87, 2024-12-11:
I tried now something new. I setup a second mqtt broker (IP 10.42.0.248)  in the same network like the RaspAP  (IP 10.42.0.1) and now the inetbox2mqtt could successfull connect. There should probably an issue with the routing of the MQTT-request. I'm not sure how to fix it. If i use my smartphone as an MQTT-Client i can successfull send an request to an broker which is installed on the same machine as the RaspAP but not with the inetbox2mqtt :(