mc0110/inetbox2mqtt #83: Heating switches off

gdm008, 2025-01-02:
Hello, I have the problem that the control sometimes no longer switches the heating on again after it has reached the target temperature.
As well as a problem that suddenly the water heater runs without me turning it on.

Saaiber, 2025-03-27:
Same problem here with a 179 euro Womolin lin controller, it works fine for one day and then it disconnects. Womolin says it is a software problem and the developers of the software are not very helpful too. I gave up....I will dismantle the womolincontroller and use the (very expensive ;) ) ESP32 for other purposes...

HBK1, 2025-03-27:
Have you tried:
https://github.com/Fabian-Schmidt/esphome-truma_inetbox

Saaiber, 2025-03-27:
I was hoping that this solution would work but I also tried it. For this one everything works fine except for the init of the controller in the Truma system. I can't see the box as 3th value in the index on the cpplus screen. I can connect to the webaddress of the controller and I can toggle the switches but everything returns with an error because the cpplus doesn't recognize the controller.

mc0110, 2025-04-01:
If you want to discuss Fabian's software, please do so in his Github channel. My software has not been developed for the Womolin Controller, the use of the software is up to you.

closed 2025-04-01 by mc0110