mc0110/inetbox2mqtt #84: No Webinterface in Normal Run

janner2000, 2025-01-21:
Hello..
Why do I no longer have a web interface after changing the mode to "normal-run" on my ESP32???

ESP32 version 265.

Sorry for the bad English, it's a Google translation.

I'm from Germany

mc0110, 2025-01-24:
Hi,
In microPython there is unfortunately a memory limitation that does not allow the web frontend and the control software to be stored in memory together. In order for the software to be executable, various parts have already been precompiled so that they do not burden the RAM area at runtime. 
In addition, the open web interface poses an unnecessary security risk, as it is not required for normal MQTT broker operation.
I hope I was able to explain it clearly.
BR
Magnus

cheand, 2025-01-24:
Hi mc0110,
i have the same problem. how can i switch the esp32 back to web interface mode if i want to change something?
Andre

unwichtig, 2025-01-24:
As i understand its only possible with an MQTT Set ... 
Would it be much work to get this switching from OS Mode with bridging an IO Port for normal Mode and an other for OS Mode?

mc0110, 2025-01-25:
Hey guys, this isn't a fault, this is a feature ;-)

With a functioning terminal connection, you can delete the control file run_mode.dat after a soft reset using the following command.
```
import os
os.remove('run_mode.dat')  
```  
After that, the chip starts again in OS mode.

closed 2025-04-01 by mc0110