mc0110/inetbox2mqtt #85: CP plus version unknown?

DarkKnight1973, 2025-02-04:
Hi there,

I followed the instructions, could flash the ESP, get to the config interface, etc. but sending commands (for testing) doesn't work (and accordingly, the overall communication with MQTT is not up and running. I have a new motor home from December 2024 and a recent CP plus which is inetready. And here is my question (issue):

- When I look up the index, I don't get anything listed starting with CP.*** 
- It has an entry with H2.00.01 (Truma Combi 6 Heating I guess) and F4.05.02 

What does the latter (F4.05.02) mean and could this be the reason why thing don't work out?

I'm now flashing and wiring another ESP just to make sure that everything works on this end.

Best, Markus

mc0110, 2025-04-01:
It would be interesting to know whether the INIT process and control is also possible with F4.05.02. Did you manage to do this?

Szort, 2025-08-02:
Currently trying to do this for CP Plus iNet Ready with F4.01.02. Unfortunetly this dose not work atm. I'll try to report any progress if any.