mc0110/inetbox2mqtt #88: Mqtt connection gone after 1 day of correct operation

Saaiber, 2025-03-12:
Hello,

I finally managed to flash my womolin lin interface, enter the needed credentials, restarted in run mode and everything worked fine in my home assistant for 1 day...suddenly I see that the truma water temp and clock info isn't available anymore and settings aren't sent to the truma panel anymore. 

I see the womolin in the network on ip 192.168.1.125, so the wifi connection is ok but I can't connect to this ip-address.

My mqtt on the cerbo gx is also reachable...

How can I fix this without having to flash the thing all over again because it is happening now twice that after 1 day of correct operation. it stops again...

Nic

mc0110, 2025-03-12:
I think, this is a mqtt-problem, no Inetbox2mqtt issue

Saaiber, 2025-03-13:
Ok, any hint on what to do to troubleshoot this issue?

Saaiber, 2025-03-16:
Ok, I understand... I will figure it out by myself.

Nic

Saaiber, 2025-03-18:
I reinstalled everything, now with mqtt

Saaiber, 2025-03-18:
I reinstalled everything yesterday, now with mqqt on the homeassistant. It worked perfectly but the next day it didn't work anymore. Can someone please help me, I purchased a womolin controller and I can't keep it working for more than one day.

Nic

Saaiber, 2025-03-31:
I understand, which is why I asked: "Do you have any suggestions on how to troubleshoot this issue?" I’m new to this and eager to learn, but simply being told "it has been tested and it works" doesn’t really help me move forward.

Initially, everything worked fine, but after a day, the connection to the controller was lost. I purchased a ready-made Womolin Lin controller, expecting it to function without issues. However, I’m now in a situation where the hardware developer points to the software as the cause, while the software maker insists that everything is working correctly.

I would really appreciate any guidance, and I hope you can also understand my perspective. I’m also willing to pay a fee for assistance in resolving this issue.