mc0110/inetbox2mqtt #89: Problem mit INIT bei der Truma

Tins612, 2025-03-21:
Hallo ich habe folgendes Problem, nachdem die ESP32 Verbindung manchmal warum auch immer getrennt wird und somit die Heizung der Truma einfach ausgeht wollte ich auf den Pico umsteigen. Ich bekomme das auch soweit hin dass ich auf die Weboberfläche des Pico komme und die Credentials pflegen kann. Er ist auch im Wlan im Wohnmobil zu sehen, aber beim Init Prozess der Truma CP Plus funktioniert es nicht mehr. Wenn ich den ESP32 wieder dran hänge klappt alles. 
Woran kann das liegen, ich habe wie in der Beschreibung Pin 4 vom Pico auf den RX vom TJA1020 angeschlossen und den Pin 5 auf den TX. GND ist auch angeschlossen. Die Spannungsversorgung des Pico habe ich erstmal über ein USB Ladekabel hergestellt. Ich weiß nicht weiter

Tins612, 2025-03-25:
Hello, I have the following problem: After the ESP32 connection sometimes disconnects for whatever reason, causing the Truma heater to simply turn off, I wanted to switch to the Pico. I've managed to get to the point where I can access the Pico's web interface and manage my credentials. It's also visible on the Wi-Fi in the motorhome, but it no longer works during the Truma CP Plus initialization process. If I reconnect the ESP32, everything works fine.

What could be causing this? As described, I connected pin 4 of the Pico to the RX of the TJA1020 and pin 5 to the TX. GND is also connected. I initially connected the Pico's power supply via a USB charging cable. I don't know what to do next.

Deleny74, 2025-03-28:
Hello, I have exactly the same problem. My ESP32 kept crashing for no apparent reason, so I wanted to switch to the rp pico W 2040 as recommended. The installation went smoothly, and I can use the pico in OS mode and RUN mode. Also I have a connection to my mqtt broker. 
Unfortunately, the Truma CP+ doesn't find the pico at the init prozess. When I connect the ESP32, initialization works immediately. To rule out anything wrong with the pico, I first changed the GPIOs, then ordered and tested a new pico. It's always the same problem. The pico can't be found after initialization. Do you have any other tips for me?

>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
INFO:__main__:release no: 2.6.5
HW:  RP2
INFO:connect:Detected micropython 1.24.1. 4870 on port: rp2
INFO:__main__:Normal mode activated - for chance to OS-mode type in terminal:
INFO:connect:DHCP: True
INFO:connect:fixedIP: None
INFO:connect:LAN: False
..........INFO:connect:Wifi connected: 192.168.xxx.xxx
INFO:connect:Found Credentials
INFO:connect:MQTT Port is switched to port: xxxx
INFO:connect:Start mqtt connect task
>>>import os
>>>os.remove('run_mode.dat')
INFO:main1:HW-Check RP2 rp2
INFO:main1:prefix: 'truma' set: service/truma/set/ rec: service/truma/control_status/
INFO:mqtt_async2:Connecting to ('192.168.xxx.xxx', xxxx) id=b'e6614104033c3035' clean=1
INFO:mqtt_async2:Connecting to ('192.168.xxx.xxx', xxxx) id=b'e6614104033c3035' clean=0
INFO:connect:Interface connected: 192.168.xxx.xxx
INFO:connect:MQTT connected
INFO:main1:lin-loop is running
INFO:main1:del ha_autoconfig completed
INFO:main1:set ha_autoconfig completed
INFO:main1:Initializing completed
INFO:main1:inet-key:target_temp_water value: 0
INFO:lin:watchdog activated

mc0110, 2025-03-28:
Sorry - the software has been tested very intensively and has worked stably in various configurations. I use a Pico in my RV, which has now been running for 6 months without interruption. I hope for your understanding, due to the totally different environments and areas of application, I cannot provide any assistance from a distance with minimal information

Tins612, 2025-03-28:
What information is needed?

mc0110, 2025-03-28:
What do you think? Any ideas

Tins612, 2025-03-28:
Whether I should provide any information so that they can possibly find the error. Unfortunately, I don't know any more.

Tins612, 2025-03-30:
Can I somehow change the yaml file and send it to the Pico? Maybe the settings don't fit me. I have a Truma Combi 4. Or does nothing need to be changed?

mc0110, 2025-04-01:
This all sounds like an electrical problem with the level converter. There were also some problems with the first testers. Please also have a look at the closed issues

closed 2025-04-01 by mc0110

mc0110, 2025-04-06:
Hello @Tins612, please have a look at the solution to Issue #69, we have also adapted the lin.py code. It could well be that this also solves your problems.