mc0110/inetbox2mqtt #90: Can you give me the explicit electric scheme for the Truma Duo Control

HenkDeuling, 2025-04-01:
I’ll like a similar scheme as for the leds of the MQTT and LIN status for connecting the Truma Duo Control, because I cannot figured it out how to do it with the Truma Duo Control.
Do I need only the cable from the Truma Duo Control (backside) to the Truma DuoC (the display)?

mc0110, 2025-04-01:
Unfortunately, I don't have the TRUMA Duo Control either - the inetbox2mqtt only allows 2 channels to be read out and 2 channels for switching. During the development of the software, there was a request from the community to realise this. I'm sorry, I didn't find anyone who wanted to test it at the time.

closed 2025-04-01 by mc0110

HenkDeuling, 2025-04-01:
Dear Magnus,Thanks for your quick reaction.Because I do not know much about the ESP32, I try to make the right electric connections with the Duo Control.I let you know what my progress is on this matter.Best regards, Henk DeulingVerstuurd met Gmail vanaf mijn iPad

mc0110, 2025-04-01:
The Duo-Control runs on 12V. If I remember correctly, one of the switches is for a heater. If you realise the circuit as described in the readme, then you should be able to control the 12V with the ESP.