mc0110/inetbox2mqtt #91: No Connectipn to CP Plus

DerMatze3279, 2025-04-02:
Hi all, I am trying to connect my system to the Truma panel without success.

I am using ESP 32 with UART interface. TX connected to port 15, RX to 14 and GND connected. LIN and 12V+ connected at the other end. 
RJ12 connected to the second port of the heater system (disconnected INET Box before and resetted the panel - Index only showing 2 devices).
ESP is running an connected to MQTT. No error messages. 
CP PLUS panel Version: C4.03.02

I am not getting the LIN connection up and running - resetting the panel does not solve the issue. Tested in OS run mode as well.

Any suggestions on how to get this up and running?

Thanks in advance,
Matthias

mc0110, 2025-04-03:
It sounds like an electrical problem. Did you pay attention to [the note ELECTRIC](https://github.com/mc0110/inetbox2mqtt/blob/main/doc/ELECTRIC.md)?

DerMatze3279, 2025-04-06:
Probelm solved for the LIN Connection - i had to swap TX and RX. Thanks!

mc0110, 2025-04-06:
Great news - thanks for your feedback

closed 2025-04-06 by mc0110